Bodmer/TFT_eSPI #2165: ESP32 with ST7789V3 (Waveshare LCD 1.47" 172x320 rounded corners)

washburnit, 2022-11-13:
Hi,
I'm trying your library with the display in the subject.
I loaded the "Colour test" example code, configured the "User_Setup_Select.h" enabling the "Setup72_ESP32_ST7789_172x320.h" line.
In the "Setup72_ESP32_ST7789_172x320.h" I had to change the line:
#define TFT_RGB_ORDER TFT_BGR --> #define TFT_RGB_ORDER TFT_RGB
in order to have the correct colors otherwise the line "Red text" appeared in blue and the "Blue text" appeared in red.
I had also to set "tft.invertDisplay(true)" to have a black background color otherwise "tft.fillScreen(TFT_BLACK)" painted the screen with white.
After those variations the example code worked but the display was not "entirely" drawn, there was a 172x100 (supposed) pixel area that appears with garbage (random red blue green pixels) and it's not used, it's like the real resolution were greater than 172x320.
What could I check?
Thank you, regards.

Roberto

washburnit, 2022-11-13:
I found the solution for the "partial screen" by myself...I had to select the correct display driver into the "User_Setup.h".
But I have a question anyway: is that display driver suitable also for a Raspberry Pico W?
Thank you, regards.

Roberto

Bodmer, 2022-11-13:
Glad you found the solution.

Yes, it will work on a Pico and Pico W. If you are currently using an ESP32 then when you change to a Pico then you will need to change the GPIO numbers. There are example setups for RP2040 processors in the User_Setups folder of the library.

closed 2022-11-13 by Bodmer