Bodmer/TFT_eSPI #2172: printf error

johnerb44, 2022-11-14:
using rp2040
mbed rp2040 package

flash_png.ino

Example output:

```
  Serial.begin(115200);
  tft.init();
```
](error: 'class arduino::UART' has no member named 'printf_'; did you mean 'print'?)

Bodmer, 2022-11-14:
The error seems to be reported by another library called LibPrintf. I am not sure why such a library is needed as many supported processors have printf built into the board package. Maybe you can delete that library as it does not seem to provide the full functionality expected for printf.

Which processor are you using?

johnerb44, 2022-11-15:
I am using a Raspberry Pi Pico RP2040.
Thanks for the quick reply. Sorry to waste your time. I have used your library with Esp32 and esp8266 before and really liked it. Never had a problem.
I should have investigated the LibPrintf library more. To get it to compile I needed to add printf_init(Serial); to setup()

void setup()
{
  printf_init(Serial);
  Serial.begin(115200);
  Serial.println("\n\n Using the PNGdec library");

  // Initialise the TFT
  tft.begin();
  tft.fillScreen(TFT_BLACK);

  Serial.println("\r\nInitialisation done.");
} 
Also changed Serial.printf() statement to just printf().

Without adding the LibPrintf library (there are others I could have tried) it didn't look like the mbed rp2040 package supported printf().

johnerb44, 2022-11-15:
See previous comment. Raspberry Pi Pico Also my first instinct was to use the https://github.com/earlephilhower/arduino-pico board rp2040 package but in the Arduino IDE it seems to have said it was deprecated and to use the mbed package.   I now notice that the earlephilhower package specifically says it supports printf. I think I will try it and see if your library compiles without having to use LibPrintf. Thanks again. I will close the issue. Sent from Mail for Windows From: BodmerSent: Monday, November 14, 2022 5:33 PMTo: Bodmer/TFT_eSPICc: John; AuthorSubject: Re: [Bodmer/TFT_eSPI] printf error (Issue #2172) The error seems to be reported by another library called LibPrintf. I am not sure why such a library is needed as many supported processors have printf built into the board package. Maybe you can delete that library as it does not seem to provide the full functionality expected for printf.Which processor are you using?

closed 2022-11-16 by johnerb44