Bodmer/TFT_eSPI #2182: Pixel read from ST7789 via SPI Interface

SkydiveH, 2022-11-17:
Hi, I'm using this library and are very happy with it. Really, this Lib is great. I'm using it with a F446 STM and the connection via SPI is very easy. Now I'm working on reading out the value of the pixel. I'm using the example and the value returned is always FFFF. I checked the wiring of the TFT and found, that the read input has no connection. On other display it is set to 3.3V. My datasheet says 3.3V or GND. But how do i read values frmo the TFT when I do not use the pin. My question is, do I have forgotten to connect the pin and declare it. First I thought this is handled by the SPI-Bus but I'm not so shure. The Display is a KD022QVTBD024 from Shenzen Startek. If somebody knows how to fix this issue or had this problem in the past, please reply.

Many thanks, Klaus

Bodmer, 2022-11-19:
The display must provide a SDO (data out) signal to read data and this must be connected to the processor MISO. Not all dsplays mounted on a PCB provide this signal.

The ESP32 and ESP8266 only, can read from displays that provide a bi-directional SPI data pin, typically this signal is called SDA.

closed 2022-11-19 by Bodmer

SkydiveH, 2022-11-20:
The SDO output is connected, but it didn't send the datas out. According to the ST7789V Datasheet there are 4 IM lines. IM[3..0]. On the display there are only 3. This lines are IM0 to IM2. That means on the display the IM3 is on low or high. In this display is lies on gnd and the SDA line is used as In and Out. Would IM3 be on high, the SDO would be the output. 
[image]
We tried to solve this problem with uncommenting the TFT_SD_READ. But it does not work  and you write, it is only working with the ESP32 only.
How to activate it for the STM32, here the F446RE.

The rest of my sketch is workinmg perfectly, but I need to access to the pixel data. 

Has somebody experience with this problem?

Bodmer, 2022-11-20:
You could try this circuit:
[image]

It effectively makes a half duplex bi-directional SDA connection compatible with a standard MISO+MOSI SPI port.

The 1K resistor means that both MOSI and SDA can be output and the SDA output will be able to drive the MISO line without conflict with the MOSI output. The disadvantage is the the 1K resistor causes a phase delay due to stray capacitance at the TFT input (i.e. acts like a low pass RC filter), this may require the SPI clock rate to be reduced.

SkydiveH, 2022-11-21:
Sorry for troubling you again
I had time now to test the installation of the 1k resistor,it but when I connect theMISO pin then the display goes white. When I open MISO (PA6) then after a reset everything is ok. A reset on the Nucleo board with this resistor in between does not change anything.
The manufacturer of the display has confirmed, that he has put the IM3 to ground
[image]
Then I now wonder why he has still the SDO output pin. During init of the display I have seen with the O'scope datas going out and the quiet.

Do you now an improvment to the 1k resistor.

Bodmer, 2022-11-21:
Did you reduce the SPI clock rate as suggested?

SkydiveH, 2022-11-21:
Yes, i reduced to 10 mc

Bodmer, 2022-11-21:
Post your setup file and list all connections made to the TFT and processor.

SkydiveH, 2022-11-22:
Setup file as follows
[image]
The board is nucleo 64 STM32F446RE. The connections are shown in the file above.

When I connect the MOSI via the 1k0 and connect the MISO line to the side of the processor, the the display works. That means, the PA7 and PA6 are jumpered and then it goes via the 1k0 to the SDA IN/Out on the Display. The program works but shows 0 and error.
[image]

If I change the 1k0 as it is shown in your dwg, the the screen goes white. Program runs with the same readout on the serial monitor. 0 and Error

I tried it with the SDA_READ statement and without. I found, that it is still under construction so I think I have to take it out of the ESP routine, were the SDA_READ is defined and transfer it to the STM32 part. But thats only a guess, that this might work.

Bodmer, 2022-11-24:
I am surprised it did not work. To investigate further I wired a nucleo board to a ST7789V (with bidirectionsal SDA pin) display tried the resistor based approach as suggested and it worked fine. I reduced the send frequency to 20MHz (27MHz did not work with the resistor inline due to capacitance as expected). I also reduced the SPI read frequency to 10MHz as the ST7789 required a lower read speed.  Do NOT define TFT_SDA_READ as this will make the resistor mod fail.

Here is the setup file I used with the Nucleo 446RE board:
```

        ///////////////////////////////////////////////////
        //  Setup for STM32 Nucleo and ST7789 display   //
        ///////////////////////////////////////////////////
#define USER_SETUP_ID 0x29D

#define STM32

#define ST7789_DRIVER

// My display is cropped
#define TFT_WIDTH  170
#define TFT_HEIGHT 320

#define CGRAM_OFFSET      // Library will add offsets required for cropped displays

#define TFT_SPI_PORT 1

#define TFT_MOSI D11
#define TFT_MISO D12
#define TFT_SCLK D13

#define TFT_CS   D5 // Chip select control pin to TFT CS
#define TFT_DC   D6 // Data Command control pin to TFT DC (may be labelled RS = Register Select)
#define TFT_RST  D7 // Reset pin to TFT RST (or RESET)

//#define TOUCH_CS D4

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// STM32 support for smooth fonts via program memory (FLASH) arrays
#define SMOOTH_FONT

#define SPI_FREQUENCY  20000000

#define SPI_READ_FREQUENCY  10000000

#define SPI_TOUCH_FREQUENCY  2500000
```

I have short 10cm wires between the Nucleo and display so that minimises the stray capacitance. If you use a circuit with long board traces or wired you may need to reduce the SPI write frequency further.

This is the TFT_ReadWrite_Test sketch output:
```
Pixel value written = 1
 Pixel value read    = 1
 PASS 
 Pixel value written = 2
 Pixel value read    = 2
 PASS 
 Pixel value written = 4
 Pixel value read    = 4
 PASS 
 Pixel value written = 8
 Pixel value read    = 8
 PASS 
 Pixel value written = 10
 Pixel value read    = 10
 PASS 
 Pixel value written = 20
 Pixel value read    = 20
 PASS 
 Pixel value written = 40
 Pixel value read    = 40
 PASS 
 Pixel value written = 80
 Pixel value read    = 80
 PASS 
 Pixel value written = 100
 Pixel value read    = 100
 PASS 
 Pixel value written = 200
 Pixel value read    = 200
 PASS 
 Pixel value written = 400
 Pixel value read    = 400
 PASS 
 Pixel value written = 800
 Pixel value read    = 800
 PASS 
 Pixel value written = 1000
 Pixel value read    = 1000
 PASS 
 Pixel value written = 2000
 Pixel value read    = 2000
 PASS 
 Pixel value written = 4000
 Pixel value read    = 4000
 PASS 
 Pixel value written = 8000
 Pixel value read    = 8000
 PASS 

```

Can you try this again.  Provide pictures of you wring/boards if it still does not work and I might be able to spot the cause.

SkydiveH, 2022-12-03:
Sorry for beeing late in teh response. We build up to systems and one is working with the resistor. One doesn't. I'm still looking for the failure. Up to know all is identical and no fault found. But there must be one. Test no 2 is running with the resistor. So it is working. I've ordered a display with the IM3 on VCC and hope to get it from Alibaba the next two weeks.
I keep you posted with the results of the troubleshooting. Thank you very much for always the fast response and your help.