Bodmer/TFT_eSPI #2209: XIAO Borads

jlgoasis, 2022-11-27:
Hi Bodmer.

Recently I'm using XIAO RP204 boards in my projects, as well as other XIAO.

Everything works fine choosing Raspberry PI Pico by Earle Philhower.

I know it needs some development resources from your side, but Seeed Studio XIAO boards are getting more and more popular, so I would be greatly appreciated if you would incorporate these boards in the future in your library.

Thank you in advance.

Bodmer, 2022-11-27:
The TFT_eSPI library already supports the RP2040 so it is not clear to me what is being requested.

jlgoasis, 2022-12-03:
Hi Bodmer

Sorry, let me clarify and give more details.

I am using Arduino IDE 2.0.

If I choose board from SEEED STUDIO Seed XIAO RP2040,
(https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json) when compiling I get the error message:

"/Users/jlgoasis/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:31:0:
/Users/jlgoasis/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.c:42:6: error: #error The Arduino RP2040 MBED board package is not supported when using PIO. Use Earle Philhower's community package."
     
Choosing any other Raspberry PI Pico/RP2040 board, then everything works fine.

I would greatly appreciate it if you could detect where the incompatibility is using Seed XIAO RP2040 board.

Thank you in advance.

jlgoasis, 2022-12-03:
Sorry the correct Seed XIAO RP2040 board URL is https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json

Bodmer, 2022-12-04:
I don't get that error message. It will only happen if the mbed package is used.

Tested using:
Arduino IDE 2.0.2
Philhower RP2040 package: 2.6.4
TFT_eSPI example: alphaBlend_Test
User setup: Setup60_RP2040_ILI9341.h

No error seen

Bodmer, 2022-12-05:
I suspect the board is not being selected correctly after loading the Philhower board package.

closed 2022-12-05 by Bodmer

jlgoasis, 2022-12-05:
Sorry Bodmer, the problem is not on Philhower board,  it works well, the issue is when select the Seeed XIAO RP2040.