Bodmer/TFT_eSPI #2212: Sprite Image Pixelated...

phillmj, 2022-11-29:
1. A description of the problem and the conditions that cause it to occur:

I'm having a strange issue, whereby if I push an image to the screen via tft.pushImage, it's antialiased and looks great.
If I take that same image and push it to a 16-bit Color Depth sprite, it's immediately pixelated. Same image...See attached pic.

This is an adaptation of one of your beautiful example sketches (thank-you!!!).

3. IDE (e.g. Arduino or PlatformIO): Arduino TEENSY
4. TFT_eSPI library version: 2.4.79
5. Board package version (e.g. 2.0.3) : not sure, many installed
6. Procesor, e.g RP2040, ESP32 S3 etc: TEENSY 4.0
7. TFT driver (e.g. ILI9341): 7789, Waveshare
8. Interface type (SPI or parallel): SPI

```
  void SSJustG() {

#define DELAY 1000
#define WIDTH  320
#define HEIGHT 240

  tft.fillScreen(TFT_BLACK);
  
  spr1.createSprite(WIDTH, HEIGHT);
  spr1.setColorDepth(16);
  spr1.fillSprite(TFT_BLACK);    
  tft.pushImage(133, 47, GclefWidth, GclefHeight, GClef); //<--Looks perfect!!
  delay(3000);
  
  spr1.pushImage(133, 47, GclefWidth, GclefHeight, GClef);
  spr.pushSprite(tft.width() / 2 - WIDTH / 2, tft.height() / 2 - HEIGHT / 2, TFT_BLACK);//<PIXELATED!!!

}//end void SSJustG
```

[image]
[image]

Bodmer, 2022-11-29:
This is an endianess problem. Try adding spr1.setSwapBytes(true); before pushing the image to sprites.

phillmj, 2022-11-30:
That did it! THANKS SO MUCH!!!

closed 2022-11-30 by Bodmer