Bodmer/TFT_eSPI #2222: ili4988 colours displayed wrong ESP 32

TheoSharratt, 2022-12-03:
Hiya! I've got an ili9488 8 bit parralel wired up to an ESP32.
It seemed to be running fine, but i've noticed it the colours are wrong. 
I'm not certain if its a code issue or hardware issue. 
The wires are all correct but some colours are lost or even inverted. It makes gradients rather bizarre to see. 
It happens regardless of whether the image is from a sd card jpg or being drawn from tft.fillRectHGradient(x, y, w, h, color1, color2);

Im using the Arduino IDE with the newest eSPI version

The example image is supposed to be a gradient from white to black. But instead it gets lighter and darker at random points and even breaks into purple and green at some lines.
Other weird things are blue and purple seem to get reversed and yellow and orange just merge into the same colour. 
The ring meter example code actually gets more yellow at the end of the meter instead of red
The setup is wired on a breadboard using dupoint wires. Wiggling the wires has no impact on the colours
The code used

```
  #include <SPI.h>
#include <FS.h>
#include <SD.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

// JPEG decoder library
#include <JPEGDecoder.h>

#define ILI9486_DRIVER
#define ESP32_PARALLEL_8_BIT
#define TFT_DC   17  // Data Command control pin - must use a pin in the range 0-31  (LCD_RS on my display)
#define TFT_CS   33  // Chip select control pin
#define TFT_RST  32  // Reset pin
#define TFT_RD    2
#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   21
#define TFT_D5   22
#define TFT_D6   27
#define TFT_D7   14

void setup() {
  tft.begin();
}
void loop() {

  tft.setRotation(2);  // portrait
  tft.fillRectVGradient(0, 0, 3200, 480, TFT_BLACK, TFT_WHITE);
  while (1); // Wait here
}

```
[image]

[image]

Bodmer, 2022-12-03:
Use the correct driver, replace:
#define ILI9486_DRIVER

With:
#define ILI9488_DRIVER

In the setup file.

TheoSharratt, 2022-12-03:
oh my goodness, it worked immediately. I didn't realize I had both ili9488 and ili9486 defined in the setup file. Thanks!!

TheoSharratt, 2022-12-03:
Thank you for such a quick response! I really appreciate it!!

closed 2022-12-04 by Bodmer