Bodmer/TFT_eSPI #2231: Ring Meter Slow to Update

phillmj, 2022-12-04:
4. IDE (e.g. Arduino or PlatformIO): Arduino Teensyduino TEENSY 4.0 @600MHz
5. TFT_eSPI library version: 2.4.79
6. Procesor: Teensy 4.0
7. TFT driver: ST7789
8. Interface type:SPI

1. I have the Ring Meter working with a music fader (linear potentiometer). When dragged slowly, all values output fine, but if a fader is dragged too fast, it doesn't output all values - and when I say "too fast" I don't mean very fast at all. Just above very slow.

Output range is from 0 - 127 (midi output), and sometimes the lowest value will be 11 or 7 or 5 or 1, when it should be 0. 

If I disable the meter output line, all values are output as expected as checked with a midi output monitor, but obviously the screen does not update.

I have used this same code with other graphics output, such as bar graphs, and there's no problem with all values displaying and outputting.
 
I was hoping you could look through my code here, complete code posted, and give me a suggestion as to what I can do to have the ring meter work without skipping endpoint values. I absolutely love what you've done.

Also, a side issue - the triangles are pixelated, noticeably, and I'm wondering if they can be made smooth. Not a deal breaker, but I have prided myself on making this product smooth visually, which your amazing library has supported at every turn.

Thanks so much for your expertise.

Mike

```
 //====================================================================================
//                                    VIEW LARGE RING METER
//==================================================================================== 
 
/////////////////////////////
  void ViewLargeRingMeter() {
////////////////////////////

  byte MIDIMin[4];
  byte MIDIMax[4];

  byte MinVal = 0;
  byte MaxVal = 0;

  int xpos = 36, ypos = 5, gap = 4, radius = 120;

  String strCCNameOrNumber;

  //Set the vars
  for (byte y = 0; y < FaderCount; y++) {
    
    MIDIMin[y] = SceneFaderRanges[CurrentSceneNumber][y*2];
    MIDIMax[y] = SceneFaderRanges[CurrentSceneNumber][y*2 + 1];
  }

  ScreenSaverClockStart = millis();

  while (!AButtonEventOccurred && !StartScreenSaver) {

    WithinLoop = true;
    WithinLoopCurrentTime = millis();

   if (ScreenSaverTimeNames[ScreenSaverTimeIntervalIndex] != "Never") { 
      ScreenSaverClockInterval = millis() - ScreenSaverClockStart; 
    }
    else {
      ScreenSaverClockInterval = 0;//So SS will never start...
    }
        
    if (ScreenSaverClockInterval > ScreenSaverTimeInterval) {
      StartScreenSaver = true; 
    }

     for (int y = 0; y < FaderCount; y++){             
      analog[y].update();   
      if(analog[y].hasChanged()) {    
        Data[y] = analog[y].getValue()>>3;
[... 174 lines cut ...]
  spr2.deleteSprite();  

  //Draw the CC Name or Number
  tft.setTextPadding(30*3);//Allow for three characters 10 pixels wide
  // Print units, if the meter is large then use big font 4, othewise use 2  
  tft.loadFont(NotoSansRegular36);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setTextDatum(TC_DATUM);
  if (tft.textWidth(units) < 3) {
    units = " " + units + " ";//for Blackout when changing to different CC's
    tft.setTextPadding(35*3);
  }
  if (r > 100) tft.drawString(units, xC, y + 77); // Units display
  else if (r <= 100 && r > 84) {
    tft.drawString(units, x, y + 67, 4); // Units display
  }
  else tft.drawString(units, x, y + 35, 2); // Units display

  tft.setTextPadding(0);//MUST have this here or it screws up the rest of the UI! After changing the text padding it is remembered unless you reset it.
  tft.unloadFont(); //MUST have this here, or on subsequent visits here it displays the loaded font above instead...the tft remembers fonts you load across voids!

  // Calculate and return right hand side x coordinate
  return x + r;
}//end function ringmeter
```

phillmj, 2022-12-04:
Issue resolved; I was calling another function that was slowing things down in the loops. Thanks.

closed 2022-12-04 by phillmj

Bodmer, 2022-12-05:
Good to hear you have it sorted.

There is no anti-aliased triangle function in the library. Maybe the anti-aliased wedge line function would suffice. Example sketch and function [here](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Smooth%20Graphics/Smooth_Graphics_Demo/Smooth_Graphics_Demo.ino#L84-L88).

phillmj, 2022-12-05:
Many thanks; I will check it out.

I love all the antialiased shapes you’ve added; I hope you add more.

All the best,

mike