Bodmer/TFT_eSPI #2238: ESP32-S3 TFT_eSPI blocking SPI if more than one SPI bus is in use

dsilletti, 2022-12-08:
If I use only one SPI device, a TFT or for example an SPI AFE chip, everything works fine on both HW SPI and software defined SPI.

If I use 2 SPI devices on 2 different SPI bus, e.g. for the TFT MOSI: 41, SCLK:  42, SS :38,DC : 39, RST:40 and for the 2nd device the default SPI bus (MOSI: 11, MISO: 13, SCK: 12, SS: 10) both SPI devices are not working

With the same setup if I simply replace TFT_eSPI with Adafruit ST7789 lib both SPI devices are working fine.

 -IDE  Arduino 
-TFT_eSPI library version latest
-Board package version latest 2.0.5 git master
-Procesor ESP32 S3-WROOM-1 all flavours
-TFT driver ST7789
-Interface type SPI

Bodmer, 2022-12-08:
There are 2 hardware SPI interface functions available to users (HSPI and VSPI) in the ESP32-S3 that can be allocated to pins. [Information here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_master.html). I suspect the libraries being used are both trying to map the same hardware function to different pins causing a conflict in settings. By default TFT_eSPI will use the VSPI (port 2) hardware block. If 
```
#define USE_HSPI_PORT
```
is put in the setup file then it will use the HSPI (port 3) block.

Let me know if this works.

dsilletti, 2022-12-08:
Hi, I already tried this setting but did not help.
Will do some other test, I will try also to create a new SPIClass for the 2nd device forcing to use the HSPI, or the VSPI if I #define USE_HSPI_PORT for the TFT_eSPI.
The strange thing is that it's working out of the box with the LovyanGFX and Adafruit 7735/7789 with no additional config.

closed 2022-12-08 by dsilletti

Bodmer, 2022-12-09:
Great, I can delete the "not tested" in the [setup notes]!(https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70b_ESP32_S3_ILI9341.h#L29).

dsilletti, 2022-12-09:
Yep, works good. Thanks

 FYI (not an issue) on the ESP32S3 I have not configured MISO (-1) and I receive this error in the debug:

[E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!

Bodmer, 2022-12-09:
In the Arduino world setting a pin to -1 means don't allocate a signal pin. The S3 support package does not handle this correctly and generates the warning. I don't 
think there is a solution at the moment, other than allocating a pin. This has been raised as an issue for the board package.