Bodmer/TFT_eSPI #2243: Wedge Line Doesn't Respect Z-order

phillmj, 2022-12-11:
1. The wedge line function draws on top of all graphics in a sprite, even if drawn before. Normally, earlier drawn items are "under" later drawn items within the same sprite; similar to layers in graphics programs, and sometimes this is called the "z-order". Attached is an adapted example from one of the excellent demos you have provided.
2. IDE Teensiduino
3. TFT_eSPI library version 2.4.79
5. Procesor, Teensy 4.0
6. TFT driver 7789
7. Interface type SPI

[image]

Example output:

```
 for (int angle = -HalfAngle; angle <= HalfAngle; angle += 2) {

    //MIN/MAX RANGE LIMITING LONG SPIKES
    if (MaxLimited(MasterFaderNum)) {
      getCoordRing(cx, cy, &px1, &py1, &px2, &py2, r3, r4, MaxAngle);  
      spr5.drawWedgeLine(px1, py1, px2, py2, w3, w4, GlowColor, TFT_BLACK); 
      //spr5.drawString(String(MIDIMax[CurFaderNum]), px1 + LMTextOffset + 2, py1);          
    }
    if (MinLimited(MasterFaderNum)) {
      getCoordRing(cx, cy, &px1, &py1, &px2, &py2, r3, r4, MinAngle);  
      spr5.drawWedgeLine(px1, py1, px2, py2, w3, w4, GlowColor, TFT_BLACK);
      //spr5.drawString(String(MIDIMin[CurFaderNum]), px1 + LMTextOffset + 2, py1);           
    }

    //SMALL DIVISIONS
    getCoordRing(cx, cy, &px1, &py1, &px2, &py2, r1, r2, angle);
    colour = rainbow(map(angle, MinAngle, MaxAngle, 0, 127));
       
    //Normal
    if (!RangeInverted(CurFaderNum)) {
      if (!MinLimited(CurFaderNum) && !MaxLimited(CurFaderNum)) {
        colour = rainbow(map(angle, MinAngle, MaxAngle, 0, 127)); //GlowColor  
        if (angle > ValueAngle) colour = DialTicks;
      }
      else {//range is limited
        colour = DarkHint; 
        if (angle > MinAngle) colour = rainbow(map(angle, MinAngle, MaxAngle, 0, 127));
        if (angle > ValueAngle) colour = DialTicks;
        if (angle > MaxAngle) colour = DarkHint; 
      }      
    }
    //Inverted
    else {
      if (!MinLimited(CurFaderNum) && !MaxLimited(CurFaderNum)) {
        colour = rainbow(map(angle, MinAngle, MaxAngle, 0, 127)); //GlowColor  
        if (angle > ValueAngle) colour = DialTicks;
      }
      else {//range is limited
        colour = DarkHint; 
        if (angle > MaxAngle) colour = rainbow(map(angle, MinAngle, MaxAngle, 0, 127));//GlowColor
        if (angle > ValueAngle) colour = DialTicks;
        if (angle > MinAngle) colour = DarkHint;  
      }      
    }     
    //tft.drawWedgeLine(px1, py1, px2, py2, w1, w2, colour, TFT_BLACK);
    spr5.drawWideLine(px1, py1, px2, py2, 2, colour, TFT_BLACK);

    //LARGE DIVISIONS
    //Normal
    if (!RangeInverted(CurFaderNum)) {
      if (!MinLimited(CurFaderNum) && !MaxLimited(CurFaderNum)) {
        colour = LEDbg;   
      }
      else {//range is limited
        colour = DarkHint; 
        if (angle > MinAngle) colour = LEDbg;
        if (angle > MaxAngle) colour = DarkHint;
      }      
    }
    //Inverted
    else {
      if (!MinLimited(CurFaderNum) && !MaxLimited(CurFaderNum)) {
        colour = LEDbg;    
      }
      else {//range is limited
        colour = DarkHint; 
        if (angle > MaxAngle) colour = LEDbg;
        if (angle > MinAngle) colour = DarkHint; 
      }      
    }     
   
    getCoordRing(cx, cy, &px1, &py1, &px2, &py2, r1long, r2long, angle);
    if (angle == -90 || angle == -66 || angle == -44 || angle == -22|| angle == 0 || angle == 22 || angle == 44 || angle == 66 || angle == 90) {
      spr5.drawWedgeLine(px1, py1, px2, py2, w1, w2, colour, TFT_BLACK);
      //spr5.setTextColor(colour, textBG);     
    }

    //"RPM" NUMBERS

    //Normal
    if (!RangeInverted(CurFaderNum)) {
      if (!MinLimited(CurFaderNum) && !MaxLimited(CurFaderNum)) {
        colour = DialGleam;   
      }
      else {//range is limited
        colour = DarkHint; 
        if (angle > MinAngle) colour = DialGleam;
        if (angle > MaxAngle) colour = DarkHint;
      }      
    }
    //Inverted
    else {
      if (!MinLimited(CurFaderNum) && !MaxLimited(CurFaderNum)) {
        colour = DialGleam;    
      }
      else {//range is limited
        colour = DarkHint; 
        if (angle > MaxAngle) colour = DialGleam;
        if (angle > MinAngle) colour = DarkHint; 
      }      
    }     
    
    getCoordRing(cx, cy, &px1, &py1, &px2, &py2, r1Num, r2Num, angle);
    spr5.setTextColor(colour, textBG, true);
    spr5.loadFont(RubikExtraBoldItalic16);
    spr5.setTextDatum(MC_DATUM);  
       
    if (angle == -90) spr5.drawString("0", px1, py1);
    if (angle == -66) spr5.drawString("16", px1, py1);
    if (angle == -44) spr5.drawString("32", px1, py1);
    if (angle == -22) spr5.drawString("48", px1, py1);
    if (angle == 0) spr5.drawString("64", px1, py1);
    if (angle == 22) spr5.drawString("80", px1, py1);
    if (angle == 44) spr5.drawString("96", px1, py1);
    if (angle == 66) spr5.drawString("112", px1, py1);
    if (angle == 90) spr5.drawString("127", px1, py1);

  }//end for

```

Bodmer, 2022-12-11:
The library renders directly to sprites as each function is called. There is no layers handling.

phillmj, 2022-12-12:
I don't think this should be closed, as there is a z-order to all of your sprites so far. That's how I've written 30,000 lines of code in the last year using your amazing library. Normally, items in sprites written earlier in the void print first, below items written later. The wedge line does not respect this.

I know you know your stuff 100%, I'm just saying there is an issue here.

Mike

Bodmer, 2022-12-12:
I suspect there is an issue with the sketch. The library just renders each function serially using the same lower level primitives.
Maybe HalfAngle is not a multiple of 2, then drawString will not get called. Shortage of RAM can also cause odd effects.
If you provide a complete example sketch that demonstrates the problem I will run tests.

phillmj, 2022-12-12:
By they way I used to get some really odd effects when using sprites. Finally I started deleting and recreating sprites every single time I use them, and that's solved a lot of issues for me. I know it's not related to this but I just thought I'd mention, as you'll see this throughout the void above.

Again, utterly fantastic work on your libraries!!

Mike

closed 2022-12-18 by Bodmer