Bodmer/TFT_eSPI #2244: tuch  do not respond Sometimes

shalom198, 2022-12-11:
Sometimes (quite often) when you turn on the device the touch screen does not respond to the touch, and if you turn it off and on once or twice it responds again. But if you turn it off again, it happens that it does not respond to touch until it is turned off and on again. It happens to me on all the screens I tested (dozens of 2.4 inches), it seems that there is some kind of problem with the initialization, does anyone know a solution?

Bodmer, 2022-12-11:
I have not seen this behaviour and it is not possible to deduce the root cause without detailed information on your hardware, connections and a sketch that demonstrates the problem.

shalom198, 2022-12-12:
hi thanks for the quick reply
here the schem, it is very simple 
[Schematic_MINI D1 + SCR_2022-12-12.pdf](https://github.com/Bodmer/TFT_eSPI/files/10205275/Schematic_MINI.D1.%2B.SCR_2022-12-12.pdf)

shalom198, 2022-12-12:
> It may be the power on reset is not effective. This may be because the
> power rail rises to operating voltage too slowly. Can you connect the TFT
> reset to a pin and define this in setup to test this?

ok' i will try it
alot of thanks

‫בתאריך יום ב׳, 12 בדצמ׳ 2022 ב-12:20 מאת ‪Bodmer‬‏ <‪
***@***.***‬‏>:‬

shalom198, 2022-12-12:
i will connect the reset pin to 1 of the mini D1 , start with HIGH and then
peak of LOW  and return to HIGH or opposite ?

‫בתאריך יום ב׳, 12 בדצמ׳ 2022 ב-13:31 מאת ‪SH JO‬‏ ***@***.***
‬‏>:‬

Bodmer, 2022-12-12:
If you define the pin in the setup file the TFT_eSPI library will handle the reset. No change to the sketch is required.

shalom198, 2022-12-12:
got it. I set the reset -1 in the driver
I will change it to another pin
Once again, thank you very much

‫בתאריך יום ב׳, 12 בדצמ׳ 2022 ב-13:40 מאת ‪Bodmer‬‏ <‪
***@***.***‬‏>:‬

shalom198, 2022-12-12:
Do I need to configure anything else in my software besides the line
"#define TFT_RST D5"   in the file "User_Setup.h , "  ?

‫בתאריך יום ב׳, 12 בדצמ׳ 2022 ב-13:51 מאת ‪SH JO‬‏ ***@***.***
‬‏>:‬

Bodmer, 2022-12-18:
Assumed resolved.

closed 2022-12-18 by Bodmer

shalom198, 2022-12-18:
Sorry, I was very sick. I started to check, but I had to change the PCB and I still haven't had time.