Bodmer/TFT_eSPI #2251: T_CS pin

gnlinuz, 2022-12-13:
Hi Bodmer,

Thank you for your lovely library, it was the only one that made my esp32 and tft work. Most of the examples are working fine, then I had some issues with "#warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!"
I read some of your replies and in order to avoid the above error you need to edit the lib and enable the T_CS pin. I have this tft3.5 inch "bought from amazon" that has from one side pins to power the lcd and the pins for LCD_RD LCD_WR etc and from the other side the D0-D7 pins with SD_SCK SD_DI etc for the SD-CARD but there is no such  pin T_CS.(the tft is like a shield for arduino very popular) I know that touch is working under arduino uno and some different lib but how can I enable the touch mode with your lib since there is no pin T_CS. Is there any other option to enable the touch with those tft? Thank you for your time and all your efforts!!!!! (I don't know if it is allowed to upload a photo of the shield)

gnlinuz, 2022-12-13:
[image]

Bodmer, 2022-12-14:
Warnings can be disabled by including:
```
defined(DISABLE_ALL_LIBRARY_WARNINGS)
```
in the setup file.

The screen has an resistance analogue interface which is not supported by TFT_eSPI. See [Readme](https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support)  for details of use with a third party library.

closed 2022-12-14 by Bodmer