Bodmer/TFT_eSPI #2253: smooth fonts from sd card.

Andrejkos, 2022-12-14:
Здравствуйте. Можно ли исправить артефакты, показанные на фото, при отображении гладкого шрифта из SD card.? Такие же артефакты и при отображении BMP-иконок.
Я использую плату ESP32 DEVKIT V1, TFT (320*480, SPI)  драйвер ILI9486 либо ST7796. Скетч из ваших примеров.

Hello. Is it possible to fix the artifacts shown in the photo when displaying a smooth font from an SD card.? The same artifacts are also present when displaying BMP icons.
I am using an ESP 32 DEVKIT V1 board, a TFT (320*480, SPI) ILI9486 or ST7796 driver. A sketch from your examples.

[image]
[image]

Bodmer, 2022-12-17:
This is a system level issue rather than a library issue.

There are a number of reasons this can hapen, for example some SD cards and SD libraries do not work well. A scan of the Discussions section may help as other users have solved issues with SD cards for jepg drawing.

The simplest and best option is to put the SD card on an independant SPI bus to the TFT.

closed 2022-12-17 by Bodmer

Andrejkos, 2022-12-17:
Спасибо