Bodmer/TFT_eSPI #2256: Driving a 16-bit parallel display without RD pin

afterBlink, 2022-12-17:
I'm trying to drive this [3.2" 16-bit ili9481 Arduino Mega2560/Due shield](http://www.lcdwiki.com/3.2inch_Arduino_Display-Mega2560)  from a Raspberry Pi Pico. 
Benchmark results in Serial Monitor are very impressive but the display stays white...

I wired the shield according to Setup106_RP2040_ILI9481_16bit_parallel, but don't know how to handle the _"read pin must be permanently connected to 3V3"_, as this shield appears to have no RD pin.

Here's a picture of the shield taken from the above link. The yellow indications were added by me based on some very old libraries and the Mega/Due pinouts, suggesting that they use the indicated NC pin...

[image]

This is what I've tried so far, with no luck:
- hard wire indicated NC pin to 3.3V
- indicated NC pin to GPIO28, pulled HIGH by first line in Setup()
- indicated NC pin to GPIO28, defined as TFT_RD in Setup106 file
- hard wire indicated 3.3V pin from the extra pair of pins to 3.3V
- select the HX8357B_DRIVER instead of ILI9481_DRIVER in Setup106 file
- select the HX8357C_DRIVER instead of ILI9481_DRIVER in Setup106 file
(relevant options above were tried with both '#define TFT_RD = -1' and '// #define TFT_RD')

I don't have a Mega/Due to test this (brand new) shield, but I think I'm either doing something wrong, or trying to do something impossible. Any help would be appreciated.

---
**Environment:**

- **IDE**: Arduino 1.8.19
- **TFT_eSPI library version**: 2.4.79
- **Board package version**: Raspberry Pi Pico/RP2040 2.6.4 (Philhower)
- **Processor**: RP2040
- **TFT driver**: ILI9481
- **Interface type**: 16-bit parallel
- **TFT to processor connections**: high quality short Dupont wires (no breadboard)
- **Setup file**: Setup106_RP2040_ILI9481_16bit_parallel.h (confirmed by Read_User_Setup.ino)
- **Used sketch**: /examples/480x320/TFT_graphicstest_one_lib.ino

Bodmer, 2022-12-17:
The RD pin is already tied high on those display boards. NC means "No Connect" so nothing should be connected to them, I exect they are not connected to anything on the display board anyway.

I have a similar ILI9481 display working with that setup. The HX8357B and HX8357C init sequences are not compatible with the ILI9481 and the library runs those displays at a much higher interface speed. The ILI9481 required a longer write cycle time and this is handled automaticall [in the library here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_RP2040.h#L96-L106). The cycle period is controlled at line 103, a larger number increases the cycle period, this scales with processor speed so it is assumed the CPU speed is set to 133MHz or less.

Things to try:
1. Drop the processor speed to 50MHz using the "Tools" menu option. Upload the sketch again. This will indicate whether the write cycle setting is wrong for that display (my ILI9481 stops working when the write cycle period drops below 80ns). If that works, [increase the value to 4 in this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_RP2040.h#L103), then increase the CPU speed to find the maximum that works.
2. Add extra short GND wires from the display to the processor, 4 should be sufficient. Reason: 16 data lines changing state can cause "ground bounce" which affects signal integrity.
3. Shorten connections as much as possible (my wires are ~4cm long.) Reason: long wires affect signal integrity due to "ringing" at the edges. The most critical line is the WR signal.
4. Check no interconnect mistakes have been made and there are no poor contact or internally broken wires. Yes, you may have checked this a dozen times but my display works OK!

I usually use the "Cellular_Automata" sketch to test signal integity as it bangs all 16 data lines between 1 and 0 simultaneously, so it is quite an "agressive" test.

Bodmer, 2022-12-17:
I have run my display with the latest Philhower library and it runs. There is a possibility that your display design is different. The interface connections are identical though. Here are reference picture to show what my display board looks like (This is a Pico W, but the staard Pico works too):
[image]
[image]

afterBlink, 2022-12-17:
Thank you Bodmer for your time and expertise! I'll try your suggestions tomorrow with soldered short wires and post the result.

afterBlink, 2022-12-18:
**It works!**
[image]

After switching to 50MHz and adding extra ground wires didn't work for me, I prepared for the dreaded soldering job. While the iron was still heating up, I already uploaded the suggested 'aggresive' Cellular Automata test and... that showed up!!

It turned out that the `while (!Serial);` line in  _TFT_graphicstest_one_lib.ino_ had hung it up before even initializing the display. For my first RP2040 project, I had set up a completely fresh and up-to-date environment on an Ubuntu laptop. While uploading is no problem, it looks like running programs don't 'see' a Serial port in my setup. Definitely my total lack of experience with Ubuntu, so I feel a bit embarrased for having bothered you. But then, it was the fact that you had it working that made me persist. Now it works great on 133MHz, even with Dupont wires.

Thanks for your help and for this great library, which has become nothing less than essential to my hobby.

Bodmer, 2022-12-18:
Great. Did not expect that and am puzzled by your OP which said "Benchmark results in Serial Monitor are very impressive but the display stays white..."  This indicates the graphics test was running fine but nothing showed on the screen, so something else has changed!

closed 2022-12-18 by Bodmer

afterBlink, 2022-12-18:
Yes, I realized that too after posting, The Serial results were shown on my regular Windows PC before switching to Ubuntu, so now I suspect that the extra (and shorter) ground wires made the difference.