Bodmer/TFT_eSPI #2257: Missing color while using pushtoSprite from one Sprite to another

rsramki, 2022-12-18:
Hi,
 
 I am having trouble in getting my image displayed with transparent background while pushing from one Sprite to another by using pushtoSprite option. I am not sure if i am doing anything wrong.  Image is displayed properly if Background is BLACK and issue when using any other color. 

1. IDE  Arduino framework with ESP32 Pico Dev Board and 1.8" TFT Display.
2. TFT_eSPI library version  used : 2.4.79 (Latest version) 
3. Board package version (e.g. 2.0.5)  ESP32
4. TFT driver ST7735 
7.  Interface type : SPI 

img is main background sprite full panel size
main_icon_1 is sprite 48px by 48px to display image on the screen.  (Have same issue with multiple icon sprites) 
....
     main_icon_1.fillSprite(TFT_BLACK);
     main_icon_1.pushImage(0,0,48,48,new_setup2);
     main_icon_1.pushToSprite(&img,10,10,TFT_BLACK);
 ```
Image Screenshot of issue Below 
![tft_espi_pushtosprite_issue](https://user-images.githubusercontent.com/32425707/208286870-a79bac4e-1efa-4843-b9f3-907144b6fb21.jpg)

Would Appreciate any suggestion or feedback to resolve my issue.!

Bodmer, 2022-12-18:
Check the image pixels you want to be transparent are actually back (pixel value of 0.

If that does not help then post a complete sketch that demonstrates the problem. To do this create a zip file of the colplete sketch and drop in the dialogue window.

rsramki, 2022-12-18:
Attached Complete Sketch..
When we use any other Color in Pushto Sprite its coming properly. Also without PushtoSprite its coming perfect. see other images next to this image.

[SpriteIssue-RAMKIv1.11.zip](https://github.com/Bodmer/TFT_eSPI/files/10254307/SpriteIssue-RAMKIv1.11.zip)

rsramki, 2022-12-18:
Adding the image which was converted to c file using : Generated by   : ImageConverter 565 Online
http://www.rinkydinkelectronics.com/t_imageconverter565.php

[image]

Bodmer, 2022-12-18:
I assume you are not execting the odd pixels in the icon to be visible:
[image]

These are there because you have specified the transparent colour to be black (pixel value 0x0000) but the extra spurious pixels are not black, you can see in the arrays (for example static const uint16_t new_setup2[2304]) that some pixels have a value of 0x0020, so since this is not black (actually a very dark green)  they will be rendered. Since most of these odd pixels have a value of 0x0020 you could do a text search and replace to make them 0x0000, however I expect there are other values that you will also need to correct.

[image]

closed 2022-12-18 by Bodmer

rsramki, 2022-12-19:
Thanks for the closure. Would like to know best conversion tools for bmp or jpg or png to RGB convertor. Looks like all tools are creating the same problem. Image before conversion is 100% accurate and during conversion it adding spurious colors. (Online Tools). Any suggestions are welcome.

Bodmer, 2022-12-19:
Use images that have never been compressed with a lossy compression method like jpeg (which tweaks pixel colours slightly).

Some image formats are "lossless" so the pixel values is the original master image are not changed. Another approach is to spot the duff pixels and replace with 0x0000 or use a paint program to repaint the image background.