Bodmer/TFT_eSPI #2287: ILI9488 Pico - Colours wrong with PlatformIO!

AndrewCapon, 2023-01-01:
Hi Guys,

I'm banging my head against the wall here.

If I run TFT_graphicstest_one from the Arduino IDE everything works fine.

The same code with PlatformIO produces incorrect colours, everything else looks fine.

I messed about a bit and found that if I rotate the colour 1 bit to the left everything works:

```
uint16_t rotateLeft(uint16_t u) {
  return (u << 1) | (u >> (15));
}

void setup() {
  tft.init();

  tft.fillScreen(rotateLeft(TFT_RED));
  tft.fillScreen(rotateLeft(TFT_GREEN));
  tft.fillScreen(rotateLeft(TFT_BLUE));
}
```

Has anyone got any idea of what might be going on here?

Cheers

Andy

Bodmer, 2023-01-01:
I suspect this may be caused by different compile optimisation levels or CPU clock frequency settings are affecting the bus timing.
Try reducing the SPI clock speed.

AndrewCapon, 2023-01-01:
Hi,

Thanks for the reply.

Sorry I should have mentioned I had already tried reducing the SPI speed, no difference. Well apart from the screen stops working totally if you go too low!

I will attach a logic analyser to it and grab an image of the signals...

Bodmer, 2023-01-02:
Run the "Read_User_Setup" sketch amd check the serial monitor output settings are the same for both the Arduino and PlatformIO environments.

AndrewCapon, 2023-01-03:
Hi @Bodmer,

They are the same, well they would be if the serial monitor worked with picoprobe in PlatformIO! So I checked through in the debugger.

To be honest I have had enough of this screen, it is slow as hell anyway.

Thanks for all the interest and help but I think its going in the bin and I will close this so it is not getting in your way.

All the best

Andy

closed 2023-01-03 by AndrewCapon

Bodmer, 2023-01-03:
OK, I don't have an explanation. I have not come across different behaviours like this between the Arduino and Platform IO IDEs, when the environment settings are otherwise the same. At least you have it working in the Arduino environment and it may be useful for projects that don't need fast screen updates.

AndrewCapon, 2023-01-03:
It is very weird :)

I originally looked at your Arduino code as the code I had knocked up for the pico suffered from the same colour issue, so I then got it going in PlatformIO so I could debug it and see what was going on differently. Annoyingly in PlatformIO I had the same problem! Once I found out that rotating in your code fixed the issue I tried it in my original code and it also fixed the issue. The weird part is all the register commands seem to work fine.

I have another 9488 screen on the slow boat from china, I'm betting that will not have the same problem when it arrives.