Bodmer/TFT_eSPI #2288: pushToSprite changed sprite color

jkrijger67, 2023-01-01:
Hello,

I have a problem with pushToSprite. I created a sprite with an yellow filled circle. I Pushed this to the background sprite and pushed this one to Display. On Display this circle is purple colored.

Arduino IDE 2.0.3 
tft_espi 2.4.71, the others didn't work with Lilygo S3.
Lilygo S3 ESP32 S3
Display ST7789 

Thank you.
[colorproblem.odt](https://github.com/Bodmer/TFT_eSPI/files/10328854/colorproblem.odt)

Bodmer, 2023-01-01:
If you push the sprite with the yellow filled circle direct to the screen, is it yellow?

Have you used setSwapBytes at all?

jkrijger67, 2023-01-02:
Good morning,
   
  
    
   
  
   yes, when I put it direkt to screen it is Yellow.
   
  
   I have coded setSwapBytes to TFT and both sprites.
   
  
    
   
  
   I attached a document with code and picture.
   
  
    
   
  
   Johannes.

Bodmer, 2023-01-02:
Run this sketch and tell me if it works, it should show two yellow circles:
```
#define IWIDTH  51
#define IHEIGHT 51

#include <TFT_eSPI.h> // Graphics and font library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

TFT_eSprite fgs = TFT_eSprite(&tft);
TFT_eSprite bgs = TFT_eSprite(&tft);

void setup(void) {
  tft.init();
  tft.fillScreen(TFT_BLUE);

  fgs.createSprite(IWIDTH, IHEIGHT);
  fgs.fillSprite(TFT_BLACK);
  fgs.fillCircle(1 + IWIDTH/2, 1 + IHEIGHT/2, IHEIGHT/2, TFT_YELLOW);
  fgs.pushSprite(0,0);

  bgs.createSprite(IWIDTH, IHEIGHT);
  bgs.fillSprite(TFT_BLACK);
  fgs.pushToSprite(&bgs, 0, 0);
  bgs.pushSprite(10 + IWIDTH, 0);

}

void loop() {
}

```

jkrijger67, 2023-01-03:
Hello,
   
  
    
   
  
   this is working fine.
   
  
    
   
  
   I think I have now learned how to use "setSwapBytes".
   
  
    
   
  
   Thanks Johannes.

closed 2023-01-03 by Bodmer