Bodmer/TFT_eSPI #2294: ESP32 + ILI9341 - PARALLEL_INIT_TFT_DATA_BUS was not declared in this scope

UnforeseenOcean, 2023-01-03:
1. Under all ESP32 platform, regardless of what is done,  the sketch will not compile with the error "PARALLEL_INIT_TFT_DATA_BUS was not declared in this scope".
2. Arduino IDE 1.8.13 was used.
3. TFT_eSPI library version is 2.4.79.
4. Board package version is 2.0.3 and the suggested "platform.local.txt"
5. ESP32-WROOM-32 (DOIT Esp32duino / Wemos D1 R32) was used. Both platforms (generic module and board itself) have been selected without success.
6. ILI9341 (MCUFriend, unknown vendor) and Simkeim SPFD5408 displays were used, both with no success.
7. Parallel 8-bit interface was used.

Plus further information as appropriate to the problem:
1. A direct plug-in was used as this is an Arduino-compatible board with an Arduino shield.
2. (See the code section)
3. (Ditto)
4. See the screenshot section

Screenshot:
[image]

The sketch:
[eSPI_TFT_Test.zip](https://github.com/Bodmer/TFT_eSPI/files/10335209/eSPI_TFT_Test.zip)

There is a 75% chance I missed something in the code and the library is not at fault, however I tried everything I could and yet I couldn't make it work. Any help would be appreciated.

Bodmer, 2023-01-03:
It appears this method of adding the tft setup information in the sketch only works in certain circumstances. I cannot see a work-around for this situation, it appears to be due to the way the Arduino IDE isolates the libraries and sketches during the compile process and this cannot be easily changed.

Since there are a number of problems cropping up with this apporach I will be deleting this capabilty from the library.

In this case it is necessary to use the original approach of calling up the required setup file in the User_Setup_Select.h file, this can be found within the library folder.

Bodmer, 2023-01-03:
I have removed the example and updated the ReadMe as noted above. The adding of a setup file in the sketch folder is now deprecated. If a reliable wor-around can be found this will be documented at a future date.

Thansk for reporting this.

closed 2023-01-03 by Bodmer