Bodmer/TFT_eSPI #2297: RP2040 in 8 bit parallel mode does not work with SSD1963 display

Bodmer, 2023-01-04:
### Discussed in https://github.com/Bodmer/TFT_eSPI/discussions/2283

<div type='discussions-op-text'>

<sup>Originally posted by **Technikfreak2002** December 30, 2022</sup>
Hello! 
First of all, thank your awesome library!
I have some trouble with a SSD1963 Display. It works fine with an ESP32, 8 bit mode, SSD1963_DIVERALT, and RGB order.
But if i use another MCU, in this case a Pi Pico, i get only wired colors and pixel garbage on the screen.
I also tried some STM32 boards, with no luck. STM32F103, STM32F411 (Blue and Black Pill) Nucleo64 L476RG and Nucleo144 H743ZI2. 
for every test i made a user setup to set the correct pins. but no luck.
Other display drivers, like the ILI9486, work with the pico and at with the nucleo64.

Can some give me a hint? am i missing a bit?

Thank you very much and a happy new year! :D
</div>

Technikfreak2002, 2023-01-15:
I tested the new define but it doesn't work.

I use Arduino IDE 1.8.19, under Linux Mint 21.
Earle's package at 2.6.5

Setup ist 103, but RST on 27, #define SSD1963_800ALT_DRIVER  added, #define TFT_RGB_ORDER TFT_RGB added and #define RP2040_PIO_CLK_DIV 3 added

Display is a 7 inch from Amazon. Conected via 10cm Jumper wires to a Pico Breadboard Demo board.

Sketch is the UTFT_DEMO_480x320

Thank you for your time and feedback! :D

Technikfreak2002, 2023-01-15:
[image]

Here is the setup.

Bodmer, 2023-01-15:
How have you wired the TFT_CS and TFT_RD pins?

Technikfreak2002, 2023-01-15:
CS to GND
RD to 3.3V

Bodmer, 2023-01-16:
Fixed in master library.

Display should work with #define RP2040_PIO_CLK_DIV 1

The DIV value may need to be increased if you overclock the RP2040 but mine works up to 200MHz with DIV set to 1!

closed 2023-01-16 by Bodmer

Technikfreak2002, 2023-02-04:
I tested the new version. And it works! Thank you so much!

commit c6d600b4 (2023-01-16): Fix #2297

The SSD1963 requires 18 bit colour in 3 bytes when an 8 bit parallel interface is used.

Added new PIO parallel code