Bodmer/TFT_eSPI #2301: tft.init() generates "gpio_set_level(226): GPIO output gpio_num error" for T-DISPLAY hardware

pabloandresm, 2023-01-04:
For hardware T-DISPLAY, when doing tft.init() generates the following error on serial:
E (228) gpio: gpio_set_level(226): GPIO output gpio_num error

Tracking down the problem, I've found that the Setup25_TTGO_T_Display.h defines "TOUCH_CS" to "-1". So when TFT_eSPI::initBus() is called, it executes "pinMode(TOUCH_CS, OUTPUT);" which generates that error at runtime.

The solution is simply to comment out the "#define TOUCH_CS            -1" in the Setup25_TTGO_T_Display.h file. It just should not be defined at all.

Bodmer, 2023-01-06:
Thanks, master library has been updated.

closed 2023-01-06 by Bodmer

pabloandresm, 2023-01-06:
Thank you Bodmer.

May I suggest also inside the TFT_eSPI.cpp ::initBus() an
if (TOUCH_CS >=0) {
....
}

Because removing the define TOUCH_CS might work for T-DISPLAY, but I've seen that define in other hardware.

I've put this too in my pull request.

Cheers,