Bodmer/TFT_eSPI #2308: bgfill on settextcolor not present when using a gradient background

erikkll, 2023-01-08:
When using setTextColor, boolean bgfill can be used to overwrite text. However, when using a gradient background, only setTextColor with a single color can be used. There is no boolean for bgfill. 

I also tried text padding as per the examples, but this doesnt work for smooth fonts, i think?

Bodmer, 2023-01-09:
Have a look at the ReadMe news item 7, that may address the problem.

If not provide an example sketch that demonstrates the issue.

Bodmer, 2023-01-09:
I realised that you might be expecting the library to redraw the gradient. The library does not have that capability so the sketch would have to clear the correct area. A better approach is to use a sprite and redraw the background and text in the sprite for each update and then push to the screen.