Bodmer/TFT_eSPI #2326: ESPI 2.5.0: Color Mismatch Issue

hammikb, 2023-01-16:
1. Updated to the new version and now im having a color issue where is displaying CYAN when its lower than 50 percent instead of yellow. It was working before the update, and just wanted to bring it to your attention. Ill upload a picture if needed 
3. Arduino IDE
4. TFT_eSPI 2.5.0
5. T-Display S3
6. ESP32 S3

//Battery Meter
  spr.drawRect(243, 5, 73, 25, TFT_BLACK);
  if (batteryPercentage >= 0 && batteryPercentage < 12.5) {
    spr.fillRect(245, 7, 8, 21, TFT_RED);
  } else if (batteryPercentage >= 12.5 && batteryPercentage < 25) {
    spr.fillRect(245, 7, 16, 21, TFT_RED);
  } else if (batteryPercentage >= 25 && batteryPercentage < 37.5) {
    spr.fillRect(245, 7, 24, 21, TFT_ORANGE);
  } else if (batteryPercentage >= 37.5 && batteryPercentage < 50) {
    spr.fillRect(245, 7, 32, 21, TFT_YELLOW);
  } else if (batteryPercentage >= 50 && batteryPercentage < 62.5) {
    spr.fillRect(245, 7, 40, 21, TFT_GREEN);
  } else if (batteryPercentage >= 62.5 && batteryPercentage < 75) {
    spr.fillRect(245, 7, 48, 21, TFT_GREEN);
  } else if (batteryPercentage >= 75 && batteryPercentage < 87.5) {
    spr.fillRect(245, 7, 56, 21, TFT_GREEN);
  } else if (batteryPercentage >= 87.5 && batteryPercentage <= 100) {
    spr.fillRect(245, 7, 69, 21, TFT_GREEN);
  }

Bodmer, 2023-01-16:
This indicates red and blue are swapped on your display. Change the option in your setup, to RGB or BGR, see here:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup206_LilyGo_T_Display_S3.h#L9-L10

closed 2023-01-16 by Bodmer

hammikb, 2023-01-17:
Thank you so sorry for wasting your time on that i totally read that when i was updating and it just went over my head

Bodmer, 2023-01-17:
Which setting worked for you? Maybe the default was wrong?

I don't have one of these displays so am unable to cjeck.

hammikb, 2023-01-17:
It was wrong in the file you stated, Thank you!

commit d6064fe1 (2023-01-17): Fix #2326