Bodmer/TFT_eSPI #2327: ST7789 driven by ESP32-S2 in parallel mode fails `TFT_ReadWrite_Test` diagnostic

chris-at-redfernsolutions, 2023-01-16:
## Issue

We are trying to use [this](https://www.alibaba.com/product-detail/1-3-1-3inch-1-3_62045571038.html?spm=a2756.order-detail-ta-ta-b.0.0.5e9a2fc2rBkMkL) ST7789-based display on a custom board based around the ESP32-S2. The diagnostic `examples/Tests and diagnostics/TFT_ReadWrite_Test` fails to read data back.

## Wiring
Our wiring is like so:
[image]
[image]

Note that our connector is flipped 180°, so pin **24** on our schematic corresponds to pin **1** on the datasheet:

| Name      | `LEDA` | `LEDK` | `GND` | `VDD-2V8` | `VDDIO` | `IMI/2` | `RESET` | `CS` | `D/C` | `WR` | `RD` | `SDA` | `DB0` | `DB1` | `DB2` | `DB3` | `DB4` | `DB5` | `DB6` | `DB7` | `TE` | `NC` | `GND` | `GND` |
| --------- | ---- | ---- | --- | ------- | ----- | ----- | ----- | -- | --- | -- | -- | --- | --- | --- | --- | --- | --- | --- | --- | --- | -- | -- | --- | --- |
| Our board | 24   | 23   | 22  | 21      | 20    | 19    | 18    | 17 | 16  | 15 | 14 | 13  | 12  | 11  | 10  | 9   | 8   | 7   | 6   | 5   | 4  | 3  | 2   | 1   |
| Datasheet | 1    | 2    | 3   | 4       | 5     | 6     | 7     | 8  | 9   | 10 | 11 | 12  | 13  | 14  | 15  | 16  | 17  | 18  | 19  | 20  | 21 | 22 | 23  | 24  |

The connector orientation is correct because we can manually GPIO the backlight to turn it on:

```C++
digitalWrite(PIN_LCD_PWR_EN, 0);        
digitalWrite(PIN_LCD_K, 0);
digitalWrite(PIN_LCD_A, 1);
```

## Configuration

Our firmware versions are:

| Tool | Version |
|---|---|
|Arduino IDE | 2.0.3 |
|Arduino-ESP32 core| 2.0.5 |
|`TFT_eSPI`|2.4.79|

We only have access the 'parallel' (8080) IO mode, because `IMI/2` is permantently tied to ground. Thus, our `User_Setup.h` contains:

```C++
#define USER_SETUP_INFO "User_Setup"

#define TFT_PARALLEL_8_BIT
#define ST7789

#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // ST7789 240 x 240

#define TFT_CS   13  
#define TFT_DC   12  
#define TFT_RST  14  

#define TFT_WR    11  
#define TFT_RD    10  

#define TFT_D0   9  
#define TFT_D1   8  
#define TFT_D2   7  
#define TFT_D3   6  
#define TFT_D4   5
#define TFT_D5   4
#define TFT_D6   3
#define TFT_D7   2

#define LOAD_GLCD   
#define LOAD_FONT2  
#define LOAD_FONT4  
#define LOAD_FONT6  
#define LOAD_FONT7  
#define LOAD_FONT8  
#define LOAD_GFXFF  
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

```

## Testing
Attached are our `.ino` and `User_Setup.h` files: [example.zip](https://github.com/Bodmer/TFT_eSPI/files/10429977/example.zip)

The sketch `examples/Tests and diagnostics/Read_User_Setup` gives the following output:

```
TFT_eSPI ver = 2.4.79
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 778B
Display width  = 240
Display height = 240

TFT_CS   = GPIO 13
TFT_DC   = GPIO 12
TFT_RST  = GPIO 14
TFT_WR   = GPIO 11
TFT_RD   = GPIO 10

TFT_D0   = GPIO 9
TFT_D1   = GPIO 8
TFT_D2   = GPIO 7
TFT_D3   = GPIO 6
TFT_D4   = GPIO 5
TFT_D5   = GPIO 4
TFT_D6   = GPIO 3
TFT_D7   = GPIO 2

Font GLCD   loaded
```

The sketch `examples/Tests and diagnostics/TFT_ReadWrite_Test` gives this output:
```
Pixel value written = 80
Pixel value read    = 0
ERROR                 ^^^^
Pixel value written = 100
Pixel value read    = 0
ERROR                 ^^^^
```

When powered on while running the `TFT_ReadWrite_Test`, the LCD backlight initially turns on showing a grey screen. The display then subtly alternates in brightness about 10 times over the course of 2 seconds. This suggests that the display is at least receiving *some* commands!

### Debugging attempts

Changing `tft.fillScreen(TFT_BLACK);` to `TFT_RED` or `TFT_WHITE` has no discernible effect.

* As suggested  in [this thread](https://github.com/Bodmer/TFT_eSPI/issues/557), we tried switching from `ST7789_DRIVER` to `ST7789_2_DRIVER` in `User_Setup.h`. This had no effect.

* As per [this thread](https://github.com/Bodmer/TFT_eSPI/issues/1326), we attempted:
    * Switching from `ST7789_Driver` to `ILI9486_DRIVER` [no effect]
    * Inserting `tft.begin()` after `tft.init()` in our `.ino` sketch [no effect]
    * Enabling `TFT_SDA_READ` [no effect]

___

Additionally, is there extra work required to make the library work with an `ESP32-S2`, or is it automatically detected? I noticed that some parts of the library are specifically written for `ESP32-C3` / `ESP32-S3` etc. but there is little mention of the `S2`.

Bodmer, 2023-01-17:
The S2 has been tested with for SPI displays only.

The original ESP32 and ESP32 S3 can be used with 8 bit parallel screens.

Bodmer, 2023-01-17:
I will update the ReadMe with a table later to show which processor and which interfaces are supported for each.

Bodmer, 2023-01-17:
I have connected up and tested 2 displays (ILI9341 and ILI9481) with an 8 bit parallel interface and they work fine. I do not have an ST7789 with a parallel interface to test but world expect it to work. This is my setup file:
```
#define ILI9341_DRIVER
//#define ILI9486_DRIVER

#define TFT_PARALLEL_8_BIT

#define TFT_RST    9
#define TFT_CS    10
#define TFT_DC    11
#define TFT_WR    12
#define TFT_RD    13

#define TFT_D0   1
#define TFT_D1   2
#define TFT_D2   3
#define TFT_D3   4
#define TFT_D4   5
#define TFT_D5   6
#define TFT_D6   7
#define TFT_D7   8

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

```

Bodmer, 2023-01-17:
I have also now run the TFT_ReadWrite_Test and it works fine too. 

The ST7789V has been tested with an ESP32 S3 with 8 bit parallel interface so I would expect your setup to work. This all suggests there is an electrical problem with your setup.

Lawrence-RedFern, 2023-01-17:
Hi Bodmer, thanks for testing this.

In the file TFT_eSPI.h, which processor did you have defined? 

```// Include the processor specific drivers
#if defined(CONFIG_IDF_TARGET_ESP32S3)
  #include "Processors/TFT_eSPI_ESP32_S3.h"
#elif defined(CONFIG_IDF_TARGET_ESP32C3)
  #include "Processors/TFT_eSPI_ESP32_C3.h"
#elif defined (ESP32)
  #include "Processors/TFT_eSPI_ESP32.h"
#elif defined (ARDUINO_ARCH_ESP8266)
  #include "Processors/TFT_eSPI_ESP8266.h"
#elif defined (STM32)
  #include "Processors/TFT_eSPI_STM32.h"
#elif defined(ARDUINO_ARCH_RP2040)
  #include "Processors/TFT_eSPI_RP2040.h"
#else
  #include "Processors/TFT_eSPI_Generic.h"
#endif

Bodmer, 2023-01-18:
This is the include file, it supports both ESP32 and S2:

#include "Processors/TFT_eSPI_ESP32.h"

closed 2023-01-18 by Bodmer