Bodmer/TFT_eSPI #2330: spr.createSprite(w,h) Setting width and height may be invalid

cykcykcykcyk, 2023-01-17:
Most of my projects use "createSite" for step-by-step drawing. However, I found that when I executed the blocking method for a long time, I used "createSprite". It may cause the width and height to be the minimum, and "pushSprite " will always be generated at (0,0).
For example, waiting for users to set wifi blocked for a long time. When the user has set up wifi, the screen needs to prompt the user for the next operation, which is easy to cause problems.
I don't know how to solve this problem.
My device is EPS8266+ILI9488

Bodmer, 2023-01-17:
This is almost certain to be due to insufficient RAM available. RAM is quite limited on the ESP8266. WiFi functions use a lot of RAM and can fragment the heap so a large block is no longer available.

CreateSprite returns a nullptr if the memory request failed so, for example, the following can be used:
```
  uint16_t* ptr = (uint16_t*)img.createSprite(width, height);      
  if (ptr == nullptr) Serial.println("Not enough RAM!");
  else Serial.println("Sprite created!");
```
You options are:
1. Use a smaller sprite
2. Use an ESP32 as it has more RAM

closed 2023-01-17 by Bodmer

cykcykcykcyk, 2023-01-17:
ohhhh ！！！think you！