Bodmer/TFT_eSPI #2341: RP2040 18-bit DMA

flaviut, 2023-01-21:
> A description of the problem and the conditions that cause it to occur

Using the RP2040 with the ILI9486 defaults to 18-bit color mode. Using hard SPI 0.

This renders incorrectly when using DMA, because DMA writes the buffer to the SPI bus without going through the color conversion:

```
    #define tft_Write_16(C)  tft_Write_8N(((C) & 0xF800)>>8); \
                             tft_Write_8N(((C) & 0x07E0)>>3); \
                             tft_Write_8N(((C) & 0x001F)<<3)
```

(sidenote: I don't understand the point in using 18-bit mode because we don't gain any precision going from 16->18 on the MCU rather than the display)

Minimal example:
<details>
<pre class="notranslate">
#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI(); // Invoke custom library

#define RED 0xF800

void setup() {
  Serial.begin(115200);

  tft.begin();
  tft.setRotation(0); // Landscape orientation, USB at bottom right
  tft.setSwapBytes(true);
  tft.invertDisplay(false);
  tft.fillScreen(0xFFFFF);

  tft.initDMA();

  auto *buf = static_cast<uint16_t *>(calloc(128 * 128, sizeof(uint16_t)));
  for (int i = 0; i < 128 * 128; i++) {
    buf[i] = RED;
  }

  // Start SPI transaction and drop TFT_CS - avoids transaction overhead in loop
  tft.startWrite();
  tft.setAddrWindow(128, 128, 100, 100);
  tft.pushPixelsDMA(buf, 100 * 100);
  tft.endWrite();
}

void loop() {}
</pre>
</details>

Applying the following change to `bool TFT_eSPI::initDMA(bool ctrl_cs)` resolves the issue:

```diff
--- a/TFT_eSPI/Processors/TFT_eSPI_RP2040.c
+++ b/TFT_eSPI/Processors/TFT_eSPI_RP2040.c
@@ -699,6 +699,8 @@
   if (DMA_Enabled) return false;
 
   ctrl_cs = ctrl_cs; // stop unused parameter warning
+  writecommand(0x3A);
+  writedata(0x55);           // 18 bit colour interface
 
   dma_tx_channel = dma_claim_unused_channel(false);
 
```

Screenshots:
[image]
[image]

> IDE (e.g. Arduino or PlatformIO)

PlatformIO

> TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu

`bodmer/TFT_eSPI@^2.5.0`

> Board package version (e.g. 2.0.3) available from the Boards Manager... menu

N/A

> Processor

RP2040

> TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.

ILI9486, ['2.8" TFT with Touch'](https://www.aliexpress.us/item/2255801096152347.html)

> Interface type (SPI or parallel)

SPI

Bodmer, 2023-01-22:
Yes, there is not point in using 18 bit if the display supports 16 bit in SPI mode. I have not tested the ILI9481 or ILI9486 in SPI mode, so assumed that they are like the ILI9488 which (despite what the datasheet says) only supports 18 bit mode with SPI.

A patch is also needed in Use_Setup_Select.h[ here ](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L185)as that is where the 18 bit interface code gets selected for that display type.

I will update the library now I know that 16 bit will work OK. Thanks for testing.

closed 2023-01-22 by Bodmer

flaviut, 2023-01-22:
I was wrong about the model number of the display driver--I guessed, it seemed to work, but the seller didn't say which it actually was. I've taken my display apart and it actually contains the ILI9341 instead.

The ILI9341 already has the correct bring-up command, my mistake.

Would it also be a good idea to add something like the following to clarify the situation for people who are using the 18-bit driver ICs?

```
#if defined (SPI_18BIT_DRIVER) && (defined (ESP32_DMA) || defined (RP2040_DMA))
  #error "DMA not supported with display requiring 18-bit processor, see issue #2341"
#endif
```

Bodmer, 2023-01-22:
Doh! I have reverted the changes!