Bodmer/TFT_eSPI #2343: Compilation errors when including wifimanager alongside tft_eSPI

random-ideas, 2023-01-22:
Description of problem:

When including the WifiManager.h project in any code using tft_eSPI (e.g. example of animated_eyes) I am receiving an error relating to the esp8266 library. I have tried the following test cases before raising this issue:

My setup:
Arduino 1.8.16 on Windows 11.
Latest version of tft_espi library,  wifimanger and the esp8266 board in the board manager. These on my OS are located at:

Using library SPI at version 1.0 in folder: C:\Users\%user%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\SPI 
Using library TFT_eSPI at version 2.5.0 in folder: C:\Users\%user%\Documents\Arduino\libraries\TFT_eSPI 
Using library WiFiManager at version 2.0.15-rc.1 in folder: C:\Users\%user%\Documents\Arduino\libraries\WiFiManager 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\%user%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266WiFi 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\%user%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266WebServer 
Using library DNSServer at version 1.1.1 in folder: C:\Users\%user%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\DNSServer 

What I've tried:

1. Compile and run example code on my setup for each separate library  - which worked fine (animated_eyes_1 in the case of tft_eSPI).
2. Include the wifimanager.h project in the examples provided in TFT_eSPI (#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager)

Issue I've observed:
When including wifimanager.h (https://github.com/tzapu/WiFiManager) in projects also using tft_espi, I receive a number of errors relating to the FS &fs call in functions, for libraries in:
C:\Users\%user%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266WebServer\ and C:\Users\%user%\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266

I am using an ST7789 on an ESP12 board (nodemcu).

I was hesitant to create this issue here, but I've only encountered this issue when including wifimanager in this library specifically and was hoping you might be able to provide some assistance with understanding why, as I honestly don't understand how tft_eSPI could lead to this set of errors. All I can think is perhaps the way that the tft_espi code is working with esp8266 might be causing issues? But I'm honestly stumped.

Any assistance would be greatly appreciated!

Bodmer, 2023-01-22:
I have tried to replicate the problem with an ESP8266 and the same library/board package versions but if I just add #include <WiFiManager.h> to the start of a sketch it still compiles fine, uploads and runs.

random-ideas, 2023-01-22:
Dang okay thanks for validating on your end, what version of IDE's and plugins are you using if you don't mind me asking?

I might just blow my whole environment away again and start again from scratch, it's possible my arduino IDE has some issue I've introduced on my end.

Bodmer, 2023-01-23:
Arduino IDE 1.8.13 (old but haven't bothered to upgrade)  and 2.0.3 (both work with ESP8266 board package 3.1.1 and NodeMCU 1.0 selected). Animated_Eyes_1 sketch unmodified. User_Setup.h unmodified.  WiFiManager 2.0.15-rc.1, TFT_eSPI v 2.5.0.

closed 2023-01-23 by Bodmer