Bodmer/TFT_eSPI #2363: Problem with tft.readPixel on MCU ESP32S2 and driver ST7735S

UkiTenzai, 2023-01-29:
## Description
Hello, I drew a picture on the screen and wanted to take a screenshot with tft.readPixel. 
Here is my code

```
  // draw
  tft.fillScreen(TFT_BLACK);
  tft.fillCircle(20,20,4,TFT_WHITE);
  tft.fillCircle(108,108,4,TFT_GREEN);
  tft.fillRect(64,64,16,16,TFT_RED);
  ...
  //generate picture and send it to PC
  for (int y=0; y<128; y++){
    udp.beginPacket(udpAddress,udpPort);
    for (int x=0; x<128; x++){
      uint16_t pixel = tft.readPixel(x,y);
      udp.printf("%d,", pixel);
    }        
    udp.endPacket();
    delay(10);
  }
```
Here are the **photo of the screen** and **the generated picture**
[image]

[image]

The circles' shape and location are right but the colors are wrong. And it doesn't show the rectangle. So I think there may be some bugs about the decode of  RGB. Perhaps in your code `Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp`, lines 1184 ~ 1191
```
  #else
    // Read the 3 RGB bytes, colour is actually only in the top 6 bits of each byte
    // as the TFT stores colours as 18 bits
    uint8_t r = tft_Read_8();
    uint8_t g = tft_Read_8();
    uint8_t b = tft_Read_8();
    color = color565(r, g, b);
  #endif
```
## Configuration

-  IDE:       Arduino 2.0.3
-  MCU:    ESP32-S2-MINI-1
-  screen:  TFT LCD (driver st7735s)

## Testing
I read issue #2327 and then run `TFT_ReadWrite_Test`, it failed as well.
[image]
So I convince that there is something wrong with the color.

## Debugging attempts
I changed SPI_READ_FREQUENCY, but it have no effect.

## Other
I have read the doc of  ST7735S and I will try to write a driver to read a pixel by micropython.  
I am a beginner of this field. Any help would mean a lot.

Bodmer, 2023-01-29:
Post your setup file so I can see how you have connected the S2 to the display.

Bodmer, 2023-01-30:
That is interesting. The display does not have a MISO pin. The S2 must be automatically switching the SPI bus to half duplex for it to work at all. The read frequency is too high, try reducing the SPI read frequency to 6MHz..

UkiTenzai, 2023-01-30:
I found that if SPI_READ_FREQUENCY < 10MHz or > 25MHz, tft.readPixel always return 0. 
But if 10MHz < SPI_READ_FREQUENCY < 25MHz, it return a wrong number, but not 0.

UkiTenzai, 2023-01-30:
I have no idea why SPI_READ_FREQUENCY is low enough, it also always returns 0.

Bodmer, 2023-01-30:
I think half duplex SPI with a bidirectional SDA signal is not going to work. I suggest you use a sprite instead as the screen is only 128x128 you can create a sprite of that size, draw graphcs in it and push to the screen. The pixel values can be read from the sprite, e.g. spr.readPixel(...).

UkiTenzai, 2023-01-31:
Yes, I have thought about using a buffer to store the data but I didn't know it is called sprite. I will try it later.

UkiTenzai, 2023-01-31:
It really works. Thanks.
[image]

Bodmer, 2023-01-31:
I have updated the library so that displays with a bidirectional SDA pin can be read without bus conflicts. The ST7735 pixel read code has been updated to correct the colour conversion (remove leading dummy bit).
I have tested with an ST7735 display. The setup file needs the following line to be included when using a display with a single bidirectional SDA pin.:
```
#define TFT_SDA_READ
```

closed 2023-01-31 by Bodmer