Bodmer/TFT_eSPI #2421: GC9A01 no working witch Waveshare RP2040-LCD-1.28

SunDUINO, 2023-02-18:
Hello.
The library works fine on any RP2040 with GC9A01, but with the WaveShare RP2040-LCD-1.28 module I can't get the image
Pins on which WaveShare connected the LCD:

#define TFT_DC 8 // In display, connect to DC
#define TFT_CS 9 // In display, connect to CS
#define TFT_SCLK 10 // In display, connect to SCL
#define TFT_MOSI 11 // In display, connect to SDA
#define TFT_RST 12 // In display, connect to RES
#define TFT_BL 25 // In display, connect to BLK

i tried many ways unfortunately i didn't get the image on this module. Oddly enough, the waveshare example works.

Bodmer, 2023-02-18:
Does the backlight LED come on when programmed with a TFT_eSPI example?
Post a complete setup file not just the pins.

SunDUINO, 2023-02-19:
No, the LED does not light up, only when I force the pin to state, but there is no image anyway.

configuration file...

#define GC9A01_DRIVER // Round TFT SPI Display 

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // GC9A01 240 x 240

//---------- Dotyczy waveshare RP2040_LCD-1.28 ---------------------------------------------------
//#define TFT_MISO 0 // RX PIN - Not Connected in display
#define TFT_DC   8  // In display, connect to DC
#define TFT_CS   9  // In display, connect to CS
#define TFT_SCLK 10 // In display, connect to SCL
#define TFT_MOSI 11 // In display, connect to SDA
#define TFT_RST  12 // In display, connect to RES
#define TFT_BL   25           // In display, connect to BLK

#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define LOAD_GLCD   
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6 
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT

//---------------------DLA Waveshare --------------------------------
#define SPI_FREQUENCY  33000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

pinout waveshare  

[image]

Bodmer, 2023-02-19:
It appears that User_Setup_Select.h is not pointing to the right setup file. Check this.

Run the "Read_User_Setup" sketch and look at the serial output. This should correspond to your setup file.

SunDUINO, 2023-02-20:
Hi

The backlight works, nothing can be seen on the lcd ...

[image]

Bodmer, 2023-02-20:
The display is connected to SPI port 1 so you need to add the following line to the setup file:
```
#define TFT_SPI_PORT 1
```

SunDUINO, 2023-02-20:
I added to the Setup61_RP2040_GC9A01.h file
however, no effect. only the backlight works

Bodmer, 2023-02-20:
OK, I am out of ideas, the display works fine for me with those pins and under the Arduino IDE.

Bodmer, 2023-02-20:
There is a user here that has the screen working. Check your setup against that users:

https://github.com/Bodmer/TFT_eSPI/discussions/2025

closed 2023-02-20 by Bodmer

SunDUINO, 2023-02-20:
Checked, it does not work on WaveShare, but it works on other Pico Pi on these pins ... probably the board is faulty, as long as uploading the example file from WaveShare works. It's writing anything else in their example and after compiling it works on Pico Pi with a separate display, and after uploading the file to this board it no longer works.

Thank you for all your help, I wrote to waveshare to replace the faulty tile. By the way your library is really great. Congratulations on a great job. Regards.

SunDUINO, 2023-02-21:
New board working :)