Bodmer/TFT_eSPI #2425: TFT_SPI_PORT type missmatch

Frogomeli, 2023-02-19:
When i tries to build with the last version of TFT_eSPI i got this error:
`error: cannot convert 'SPIClass' to 'uint8_t {aka unsigned char}' in assignment`

In Generic mode `TFT_SPI_PORT` is a `SPIClass` and in other mode it's a `uint8_t`

In file `TFT_eSPI.cpp`
```c++
  #ifdef TFT_SPI_PORT
    tft_settings.port = TFT_SPI_PORT;
  #else
    tft_settings.port = 255;
  #endif
```

In file `TFT_eSPI.h`
```c++
typedef struct
{
String  version = TFT_ESPI_VERSION;
String  setup_info;  // Setup reference name available to use in a user setup
...
uint8_t  port;       // SPI port
...
} setup_t;
```

And in file `TFT_eSPI_Generic.c`
```c++
// Select the SPI port to use
#ifdef TFT_SPI_PORT
  SPIClass& spi = TFT_SPI_PORT;
#else
  SPIClass& spi = SPI;
#endif
```

My build flags are:
```
-DUSER_SETUP_LOADED=1
-DILI9341_DRIVER=1
-DHASSPI=1
-DTFT_SPI_PORT=SPI3
-DTFT_CS=LCD_SS_PIN
-DTFT_DC=LCD_DC
-DTFT_RST=LCD_RESET
-DTFT_BL=LCD_BACKLIGHT
-DTFT_BACKLIGHT_ON=HIGH
-DTFT_BACKLINGT_V=2000
-DSPI_FREQUENCY=50000000
```

Bodmer, 2023-02-19:
The SPI port cannot be set for generic processors. Use only the default Arduino SPI port. Do not try to define TFT_SPI_PORT.

Frogomeli, 2023-02-19:
It need to be set for my hardware as it's not SPI but LCD_SPI (alias SPI3) (SeeedStudio Wio Terminal)
And it was working in old version.

Also for testing purpose i have change
```patch
-  #ifdef TFT_SPI_PORT
-    tft_settings.port = TFT_SPI_PORT;
-  #else
    tft_settings.port = 255;
-  #endif
```
and it works fine but it's not the ideal fix as it break all other "processors".

c.f. Mt PR to introduce TFT_SPI_PORT for generic hardware:
https://github.com/Bodmer/TFT_eSPI/pull/1682

Frogomeli, 2023-02-19:
> OK, I see the problem. I will work out a solution, in the meantime comment out the line in error.

Thx, a solution can be to change the variable in the generic file
```patch
-#ifdef TFT_SPI_PORT
-  SPIClass& spi = TFT_SPI_PORT;
+#ifdef TFT_SPI_CLASS
+  SPIClass& spi = TFT_SPI_CLASS;
#else
  SPIClass& spi = SPI;
#endif
```

but i have no inspiration for the name 🤔

Bodmer, 2023-02-19:
Library updated. Thanks for raising this.

closed 2023-02-19 by Bodmer

Frogomeli, 2023-02-20:
Thx for the fix but there a other bug

```
.pio\libdeps\seeed_wio_terminal\TFT_eSPI@src-5e2f6b2ae1485c0562034473393d0863\TFT_eSPI.cpp:5954:20: error: 'struct setup_t' has no member named 'port'
       tft_settings.port = TFT_SPI_PORT;
```

The fix is to exclude also the port assignment in the `cpp` file

```patch
diff --git a/TFT_eSPI.cpp b/TFT_eSPI.cpp
index 1edf59f..f3858d3 100644
--- a/TFT_eSPI.cpp
+++ b/TFT_eSPI.cpp
@@ -5950,10 +5950,12 @@ void TFT_eSPI::getSetup(setup_t &tft_settings)
   #ifdef SPI_READ_FREQUENCY
     tft_settings.tft_rd_freq = SPI_READ_FREQUENCY/100000;
   #endif
-  #ifdef TFT_SPI_PORT
-    tft_settings.port = TFT_SPI_PORT;
-  #else
-    tft_settings.port = 255;
+  #ifndef GENERIC_PROCESSOR
+    #ifdef TFT_SPI_PORT
+      tft_settings.port = TFT_SPI_PORT;
+    #else
+      tft_settings.port = 255;
+    #endif
   #endif
```

commit 615c6da4 (2023-02-19): Fix #2425 correct typo 2423
commit 0bad8c7a (2023-02-20): Fix #2425