Bodmer/TFT_eSPI #2460: ILI9488 with ESp32s - cannot get the touch to work

b1gmans, 2023-03-02:
Hi, Wired up my ILI9488 with resistive touch (yes it is present on the front of the display).   The touch controller demo sketch seems to calibrate ok and save away to spiffs but then the read look not giving any output even when touch screen

I thin loaded the "test touch controller" - this shows I am getting responses from the touch:
when not touched :
x: 0   y:2244   z:630   (the x stays zero, y and z values fluctuate a bit

when touch:
x: can be any number is 100's to around 2,200,  y: reduces a bit and z: 1600-2500 range

so, looks like I'm getting some readings from tft.getTouchraw() but the tft.getTouch() not returning anything.

b1gmans, 2023-03-02:
Ignore / close - realised SPI_TOUCH_FREQUENCY needs to be lower 1,000,000 not 2,500,000 - now works perfectly,

closed 2023-03-02 by b1gmans

daniu-de, 2023-03-10:
Wonder how you found this out ... unfortunately it does not change anything for me