Bodmer/TFT_eSPI #2465: ESP32-S3 7" 480x800 TN works but not with TFT_eSPI

luxxtek, 2023-03-02:
I've been using TFT_eSPI for some years and it is my favorite library.
Thank you for this masterpiece!
 
I bought this [display](https://www.aliexpress.com/item/1005004952726089.html?spm=a2g0o.store_pc_groupList.8148356.1.36425cbfgbHD7m&pdp_npi=2%40dis%21EUR%21%E2%82%AC+27%2C30%21%E2%82%AC+27%2C30%21%21%21%21%21%40210321c616776212841527285ed61c%2112000031135431788%21sh) and it has some (LVGL and other) demo sketches that run and I can upload in Arduino IDE using the Arduino_GFX_Library but I would like to use the TFT_eSPI library.
The PCB is marked **8048S070** and uses the ILI9341 driver.

Arduino IDE 1.8.19
ESP32 v 2.0.5
FTF-eSPI v 2.5.0
TFT driver ILI9341

The display works with the Arduino_GFX_Library settings as below but I have not been successful to migrate these settings to the the TFT_eSPI library. 
The display has an ESP32 S3  and I tried TFT_eSPI Setup70b without success (dark display).
```
#include <User_Setups/Setup70b_ESP32_S3_ILI9341.h>    // Setup file for ESP32 S3 with SPI ILI9341
```

**Working Arduino_GFX_Library config**

```
#define TFT_BL 2
/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
//Arduino_DataBus *bus = create_default_Arduino_DataBus();

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
//Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);
// option 1:
// 7寸 50PIN 800*480
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  bus,
//  800 /* width */, 0 /* hsync_polarity */, 8/* hsync_front_porch */, 2 /* hsync_pulse_width */, 43/* hsync_back_porch */,
//  480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 2/* vsync_pulse_width */, 12 /* vsync_back_porch */,
//  1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);

    800 /* width */, 0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    480 /* height */, 0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);
  
#endif /* !defined(DISPLAY_DEV_KIT) */
```

I ran the Read_User_Setup sketch with following result:

```
TFT_eSPI ver = 2.5.0
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 35
MISO    = GPIO 37
SCK     = GPIO 36
TFT_CS   = GPIO 34
TFT_DC   = GPIO 7
TFT_RST  = GPIO 6

Font GLCD   loaded
Fo
![ESP32-8048S070-LCM-V0 2](https://user-images.githubusercontent.com/6260582/222561465-f9ad2c45-cd2e-4d48-ba2c-4b615eeeb0c0.jpg)
nt 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
```

This display and touchscreen perform very smooth with the LVGL demo but that is beyond my goal.
I want to display some static graphics and text and have some experience with the TFT-eSPI library with the Dustin Watts PCB with ILI9488 and also with the TTGO display.

schematic of the display
[image]

Any help would be very much appreciated.

Bodmer, 2023-03-03:
The library does not support that 16 bit interface protocol so will not work. That interface style is unlikely to be supported in future.

closed 2023-03-03 by Bodmer

luxxtek, 2023-03-04:
Thank you for your confirmation. This stops my quest.

echosmart792, 2024-02-07:
Did you manage to make something appear on the display? Or just the demo code that the manufacturer made available to us? Can you put here a minimal code that works on this display?