Bodmer/TFT_eSPI #2467: TFT_eSPI with WT32-SC01

leofouhy, 2023-03-04:
I'm working with the WT32-SC01 board that uses the ST7796 display driver.
I have made what I believe are the correct changes to the eSPI files to get it to use Setup201_WR32_SC01.h
When I run your "Read_User_Setup.ino" program, it  shows the 9341 display driver is being used.
See attached file:
[image]
Can you explain what files I need to edit to make this work?

Thanks for your help.

Bodmer, 2023-03-04:
Comment out or delete this line:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L30

Remove // at start of this line:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L123

closed 2023-03-04 by Bodmer

leofouhy, 2023-03-05:
Fantastic. That worked. All because I hadn't commented out the line:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L30

Thanks for your help.