Bodmer/TFT_eSPI #2488: ili9488 with ESP32 touch not working

daniu-de, 2023-03-10:
Hi,
I bought this: [https://de.aliexpress.com/item/1005004083478309.html 6](https://de.aliexpress.com/item/1005004083478309.html) ili9844 display with capacitive touch and this [https://www.ebay.de/itm/363897056547 2](https://www.ebay.de/itm/363897056547) ESP32 WROOM32D

I got the display examples to work but I cannot get the touch to work and would appreciate any help. There is no D1 diode on the back of my (red) board

My wiring is:
T_IRQ: D27
T_DO: D19
T_DIN: D23
T_CS: D21
T_CLK: D10 (but since the display is working I assume my Board is wrongly labelled - did not work with setting SCK to 10)
SDOK(MISO): NC
LED: 5V power supply with 22Ohm resistor
SCK: T_CLK - D10 see note above
MOSI(SD): D23
D/C: D2
Reset: D4
CS: D15

[code]
TFT_eSPI ver = 2.5.0
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 21

TFT_BL           = GPIO 32
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
Touch SPI frequency   = 1.00
[/code]

When I run the Test_Touch_Controller script I keep getting this lines (without touching the display):
x: 8191     y: 8191     z: 0 
touching does not change anything - these lilnes keep popping up (autoscroll on) - there are always the same values

Would appreciate any help

Bodmer, 2023-03-11:
The library only supports the XPT2046 touch controller. You will need to find and independant library for the touch functions.

closed 2023-03-11 by Bodmer

daniu-de, 2023-03-11:
Thank you so much for answering that question again ... (sorry - just did not realize that my display also uses this chipset)

For others getting here: https://github.com/Bodmer/TFT_eSPI/discussions/2347

Glad that I managed to get your library working with the touch library mentioned in the other thread