Bodmer/TFT_eSPI #2491: Library makes bluetooth audio I2S don't work.

troisiemetype, 2023-03-11:
Hello,

I am experiencing a strange problem with the library, along with a [Bluetooth A2DP audio library](https://github.com/pschatzmann/ESP32-A2DP). I can have A2DP working, I can have TFT working, but not both at the same time : from the moment where `tft.init()` is called, I don't have sound output anymore. I've [add to a similar issue](https://github.com/pschatzmann/ESP32-A2DP/issues/372) where we started to investigate what could be the problem. It seems to be not linked to memory, as the heap always show empty space. Plus, log show that the Bluetooth audio callback is called. It looks like calling `init()` on TFT causes a conflict with I2S. Both I2S ports of the ESP have been tested. The program has been tested with init from the TFT before or after Bluetooth audio, and declaration of the instance has been tested in both order too. In each case,when TFT is inited, Bluetooth audio doesn't work anymore.
It's even stranger that I have another project where the library goes along mp3 decoding + I2S output, without a problem.

Arduino (1.8) is used.
TFT_eSPI is version 2.5.21, pulled from github in the week.
ESP is version 2.0.7.
Running on ESP32.
display ST7789v3, 172x320, SPI ([this one](https://www.aliexpress.com/item/1005004616229088.html))

You will find attached both the Arduino program and the user setup for the display.
[TFT_eSPI_bluetooth_audio_receiver_setup.zip](https://github.com/Bodmer/TFT_eSPI/files/10949507/TFT_eSPI_bluetooth_audio_receiver_setup.zip)

[Bluetooth_audio_receiver.zip](https://github.com/Bodmer/TFT_eSPI/files/10949510/Bluetooth_audio_receiver.zip)

Hope you can help, or point us in the right direction.

troisiemetype, 2023-03-11:
Ok, issue solved. It happens that on this board I'm working on, the soft mute pin of the TI PCM5102A DAC is routed to esp32's GPIO19, which is the default SPI MISO pin. As my setup file doesn't define the MISO pin, I believe it must default to input (even if it's set as OUTPUT in my setup, and set to 1), hence the absence of audio output.
Several other setup files don't define MISO : n°25, 61 (which is for RP2040, which may experience the same problem on another pin ?), 136, and also 24, 135 & 203, but those three have commented out the whole SPI pinout definition.

I've set MISO to GPIO 23, which I don't use and is non connected on my board, and it solved the problem. Maybe using -1 can solve the problem too ?

Bodmer, 2023-03-12:
Great to hear it is working.

If TFT_MISO is not defined the library will automatically assign it to -1
It appears that the Espressif board package breaks the Arduino norm and then assigns a pin which it should not.

You may find that assigning both MOSI and MISO to the same pin will work but I have not tried that myself.

closed 2023-03-12 by Bodmer