Bodmer/TFT_eSPI #2492: T-Watch V2 and readRectRGB

NDC-SC, 2023-03-12:
With the TTGO T-Watch V2, the readRectRGB instruction returns values of 255 rather than the RGB values for each pixel.
The same result with a variety of screen colours.
As a check, the uint16_t val = tft->readPixel(px, py) instruction returns 65535 (which is consistent with 2^16 for 565 colour format)

The readRectRGB instruction works well with the M5Stack Core2 ESP32, to enable screen capture.

Is there a workaround for the T-Watch V2 ESP32, please ?
Thanks

Bodmer, 2023-03-13:
It looks like the MISO pin is not connected.

The display may be configured for a half duplex  interface. Ask the vendor.

If the display is configured for half duplex then adding the following line to the setup file may work:

#define TFT_SDA_READ

closed 2023-03-13 by Bodmer

NDC-SC, 2023-03-13:
Inclusion of #define TFT_SDA_READ in the twatch2020_v2 setup file did not resolve the issue.

I built the screen image as a 240x240 Sprite and pushed the Sprite to the T-Watch screen.
With the Sprite, the readPixel instruction returned the correct 565 colour value.
However, the readRectRGB instruction still returns values of 255, irrespective of the pixel colours.

Do you know why readPixel obtains a value, while readRectRGB does not ?

Bodmer, 2023-03-13:
In that case screen CGRAM reads are not supported by the hardware.

readRectRGB is for reading screen CGRAM, not sprites.

NDC-SC, 2023-03-13:
Splitting the 565 value of a Sprite pixel, obtained with the readPixel instruction, into the RGB components, which are then written to the SD card, enables capture of the T-Watch V2 screen.

The TFT-eSPI library is a great asset. Thank you.