Bodmer/TFT_eSPI #2528: TFT eSPI on ESP32-CAM

maxmeli123, 2023-03-24:
Hi Bodmer,

I still experiment your good TFT library on  AiThinker ESP32-CAM with Arduino IDE. Last library version 2.5.0.

After some attemps to get the right pins for a TFT SPI bus, and CS, DC and RST, it works. I even tried to get a camera streaming on an ILI9341, surprisely I found even a free pin to use as TS_CS so I get the touchscreen too.

It is a good start point but this is not only gold, and I've 2 problems that I cannot solve. Maybe you have some advices so I can get it  full working.... ?

I explain better ...

- Problem 1
I've tried a couple of examples sketch, almost all seem to work, apart sketch which use sprites.
By default AiThinker ESP32-CAM board definition have the PSRAM enabled (4MB) so the library try to allocate sprites on it using ps_malloc, this all the time crash ESP32 with Guru Meditation error.

Here a line of code in the sprite.h that return error:
```
      ptr8 = ( uint8_t*) ps_calloc(frames * w * h + frames, sizeof(uint16_t));
      //Serial.println("PSRAM");
```
After this I've tried to create a simple sketch that allocate 3.500.000 bytes with ps_calloc and it works well, so Ithink the issue is related to something in your library .
After that I've tried to select ESP32 Dev Board and disabled PSRAM, now all sprites worked well, but allocated to ram.
Because I want to use this to write/read from microsd the video stream, I want to put a full frame buffer on PSRAM, then use this to show on a TFT once, but cannot get it to be working.

- Problem 2
Because ESP32-CAM is very limited with pins, it only exposes HSPI pins (GPIO14-CLK, GPIO13-MOSI, GPIO12-MISO)
What I want to do is to use HSPI to connect both TFT and microsd card slot.
I do not need to use custom pins because it slow down by using SW driver, just I need to use it as HW.
I managed with SPIClass and search to do it working but I've these results:
1) if I only connect TFT in the bus it works
2) if I only connect microsd in the bus it works, I can read and write on microsd card. Here note that I've used an external microsd card slot (instead of one already on the board)  because the original slot is connected in MMC configuration and not in SPI configuration
3) if I connect both, the TFT works but microsd refuses to mount, sometime mount but there are CRC errors

I've tried to fix it by placing a pullup resistor on other CS pins (TFT and TS) just to see if sdcard is working, but without success, if I remove the TFT from bus sdcard works, if I put it on the bus sdcard refuses to work.

Here my pin definition for ESP32-CAM in the setup config, as sdcard CS pin I use GPIO0 that I put LOW (before boot) to put ESP32 in program mode, but works as normal pin after boot:
```
//  ESP32CAM  pin definition
#define TFT_MOSI 13
#define TFT_MISO 12
#define TFT_SCLK 14
#define TFT_CS   15
#define TFT_DC   2
#define TFT_RST  16

#define TOUCH_CS  0 
```
Many thanks
Max

Bodmer, 2023-03-25:
I do not know why PSRAM is not working for sprites, it works for me and others.

The SPI instance must be the same as is used by TFT_eSPI, so this should work:
```
tft.init()
if(!SD.begin(22, tft.getSPIinstance())){
    Serial.println("Card Mount Failed");
    return;
}
```
It is also a good idea to set all the chips selects high at the beginning of setup() to avoid SPI bus conflicts during initialisation.

Bodmer, 2023-03-29:
Assumed resolved.

closed 2023-03-29 by Bodmer

maxmeli123, 2023-03-31:
Hi Bodmer, 
please do not close the issue because it is not solved.

My apologies for delay on reply but I had not much time to test it because I was busy with my work.
I can test it when I have some free time, probably this week-end or tonight.

For multiple devices on the same bus (HSPI) I've tried your advices and worked by passing tft.getSPIinstance(), but the normal way by declaring an SPIClass does not work.

Using tft.getSPIinstance() I'm now able to use TFT while read/write and list files on sdcard.
I know that the library have the possibility to read directly the touchscreen using the XPT2046, but I want mantain a possibility to use the PaulStoffregen library the same way I use with Adafruit libraries, this worked with your library but not while use sdcard.

Next I will test it by passing tft.getSPIinstance() but the problem will remains, because it will nott accept this code after tft.begin() but works with tft.getSPIinstance() ? It won't to be reinitialized ?
```
SPIClass hspi = SPIClass(HSPI);

hspi.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);
pinMode(HSPI_SS, OUTPUT);  // Documentation says that we need to do manually, the Core do not do it.

 if (!SD.begin(HSPI_SS, hspi)) {  // Passing tft.getSPIinstance() works, but with hspi with same pins fails
      Serial.println("Card Mount Failed");
      return;
   }
```

For PSRAM I have to test, on all my prevoius tests disabling it all sprites worked, if I enable it ps_calloc fails and esp32 crash with Guru Meditation error.

This week-end I will test it on other esp32 boards (eg. Wemos D32 and Wemos D32 Pro) to know if the problem was ESP32-CAM, but it seem related to the library, in a simple sketch ps_calloc worked well.

I will inform you with test results.

Many thanks
Max

maxmeli123, 2023-04-02:
Hi Bodmer,

not 100% sure but may I solved the issue (ONLY FOR PSRAM).

I use GPIO16 to connect TFT reset pin (RST) and this is not good because GPIO16 is used by PSRAM when it is activated from Arduino IDE.

Now I will go to test by declare TFT_RST in the config as -1 and reset it manually, or I need to find another free pin, the only one thar remain is GPIO4 but it is connected to FLASH LED with a small mossfet.

May another option is to use GPIO33 that is connected to rear small red led in the board.

Remain open the question because after TFT initialization I can only initialize others SPI devices only by passing tft.getSPIinstance() but I cannot get it to work all with a custom SPI configuration.

At this point my question is... it is possible create a custom SPI configuration using SPIClass and initialize the TFT by passing this object on the sketch to the library, so next I can use this instance for other devices in the bus ?

Many thanks for reply
Max