Bodmer/TFT_eSPI #2533: White space is way too small for monospaced font

rebior, 2023-03-27:
1. Description
I created vlw font files from the font "Noto Sans Mono" on Linux. The space character is excluded as mentioned in the guide. I use drawString to put smooth text on the screen and all characters have equal width except for space characters, that are way smaller.

3. IDE: PlatformIO/Arduino
4. TFT_eSPI library version: 2.5.0
5. Board package esp32dev (cant't find exact version)
6. Procesor ESP32
7. TFT driver ST7735_BLACKTAB
8. Interface type SPI

The following screenshot shows the two strings ```"HOST: "``` and ```"  IP: "```, both left aligned at the same x position. I would expect for the colons to be vertically aligned (which is the case if there are only characters other than space)
[image]

I can also provide a vlw file if it helps.

Bodmer, 2023-03-28:
The vlw files don't specify a space with, so this is guessed from the font height here:
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Smooth_font.cpp#L139

In your use case gxadvance would probably work.

rebior, 2023-03-28:
> In your use case gxadvance would probably work.

This means I should set that value manually after loading the font?

rebior, 2023-03-29:
I created a pull request that provides a simple and backwards compatible solution for that issue: #2540

closed 2023-03-31 by Bodmer