Bodmer/TFT_eSPI #2538: Problem ILI9486 STM32

jotaerrec, 2023-03-29:
Hello Bodmer! I’m using Arduino IDE, the latest version of the library. The processor I’m using is STM32F1 (blue pill). The TFT controller is ILI9486 connected by 8-bit parallel. The problem I have is that I connect it, load the program and all I get is the white tft. The pin layout is:
`
#define STM32
#define NUCLEO_64_TFT
#define TFT_PARALLEL_8_BIT

#define ILI9486_DRIVER
#define TFT_PARALLEL_8_BIT

#define TFT_RST  PB5
#define TFT_CS   PB6  
#define TFT_DC   PB7
#define TFT_WR    PB8 
#define TFT_RD    PB9
  
#define TFT_D0   PA0  
#define TFT_D2   PA2  
#define TFT_D3   PA3 
#define TFT_D4   PA4
#define TFT_D5   PA5
#define TFT_D6   PA6
#define TFT_D7   PA7 `

Bodmer, 2023-03-29:
TFT_D1 is not defined in your setup?

Have you run Read_User_Setup?

jotaerrec, 2023-03-30:
Hello Bodmer! Thanks for taking the time to respond!
If pin lcd_0 is declared. I forgot to copy it in the issue, I didn't run Read_User_Setup, I didn't know it was necessary.
Thank you very much for the time!

Bodmer, 2023-03-30:
Read_UserSetup just displays the settings the compiler has used.

jotaerrec, 2023-03-30:
Hello Bodmer! What I get from the Read_User_Setup is this:
```
[code]
TFT_eSPI ver = 2.5.23
Processor    = STM32
Transactions = Yes
Interface    = Parallel
Display driver = 9486
Display width  = 320
Display height = 480

>>>>> Note: STM32 pin references above D15 may not reflect board markings <<<<<
TFT_CS   = D3
TFT_DC   = D2
TFT_RST  = D4
TFT_WR   = D1
TFT_RD   = D0

TFT_D0   = D-64
TFT_D1   = D-63
TFT_D2   = D-62
TFT_D3   = D-61
TFT_D4   = D-60
TFT_D5   = D-59
TFT_D6   = D-58
TFT_D7   = D-57

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

[/code]
```

What should I do with this information? Thank you so much!

Bodmer, 2023-03-30:
The report looks OK.

Have you connected both 3.3V and 5V from the processor to the display to power it?

jotaerrec, 2023-03-30:
Yes, everything is well connected. I really don't know what could be the error

jotaerrec, 2023-03-30:
When executing colour_test in the serial monitor I receive this, I don't know if it will help or indicate something:
```
Pixel value written = 10
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 20
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 40
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 80
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 100
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 200
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 400
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 800
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 1000
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 2000
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 4000
 Pixel value read    = 6868
 ERROR                 ^^^^
 Pixel value written = 8000
 Pixel value read    = 6868
 ERROR                 ^^^^

```

Bodmer, 2023-04-02:
Check all your hardware connections. Use short wires. I do not have access to a hardware test environment at the moment but will run tests next week.

Bodmer, 2023-04-02:
Note that the blue pill boards must be powered via USB otherwise 5V is not present.

jotaerrec, 2023-04-04:
Yes, I have the 5v connected via USB, but I still don’t get any results. I’ll wait and see if you can test it. Thank you very much Bodmer!

Bodmer, 2023-04-13:
Delete the line:

#define NUCLEO_64_TFT

And try again.

Bodmer, 2023-04-19:
Assumed resolved.

closed 2023-04-19 by Bodmer