Bodmer/TFT_eSPI #2541: ST7789 320x240 ESP32

DrainRoot, 2023-03-29:
ST7789 SPI display, no CS pin, 320x240
The field remains 240x240 whatever you set in the settings!
Setup file Setup24_ST7789.h

// ST7789 240 x 240 display with no chip select line
#define USER_SETUP_ID 24

#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 320

//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

//#define TFT_INVERSION_ON
#define TFT_INVERSION_OFF

// DSTIKE stepup
//#define TFT_DC    23
//#define TFT_RST   32
//#define TFT_MOSI  26
//#define TFT_SCLK  27

// Generic ESP32 setup
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    -1 // Not connected
#define TFT_DC    17
#define TFT_RST   4  // Connect reset to ensure display initialises

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
//#define TFT_CS   -1      // Define as not used
//#define TFT_DC   PIN_D1  // Data Command control pin
//#define TFT_RST  PIN_D4  // TFT reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1    // TFT reset pin connect to NodeMCU RST, must also then add 10K pull down to TFT SCK

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS

Bodmer, 2023-03-31:
Run the Read_User_Setup and check screen resolution is correct.

Terryon5Star, 2023-04-05:
Bodmer can you give me a little tutorial for that 
I checkt everything but it dont work 
Or a quick Youtube Video

Discord: Trydener on 5$tar#110

Bodmer, 2023-04-06:
Post a link to the display vendors web page.

Terryon5Star, 2023-04-06:
Can u give me a link

ea7kir, 2023-04-06:
Product page: https://www.lilygo.cc/products/t-display-s3

But LilyGo only say it's an ST7789 (with trailing letters).

Sitronix appear to make the control chip:

https://www.sitronix.com.tw/en/index/

Waveshare appear to make the screen:

https://www.waveshare.com/w/upload/a/ae/ST7789_Datasheet.pdf

Terryon5Star, 2023-04-07:
Actually I mean the link of the Web Page of vendors.

Bodmer, 2023-04-07:
@Terryon5star, please raise a separate discussion and let @DrainRoot reply with the display details whish are different to the S3 parallel board.

Terryon5Star, 2023-04-09:
ok thanks

Bodmer, 2023-04-10:
Assumed resolved.

closed 2023-04-10 by Bodmer

Terryon5Star, 2023-04-10:
Bodmer I'm sorry to bother you but that is not my issue,
I have a ESP32 and a SPI 2,4 inch Display with SDCard and Touch and I don't
know how to connect it.