Bodmer/TFT_eSPI #2554: ESP32 crashes when using pushSprite inside a class method

dfreitas-git, 2023-04-03:
This is code running on an ESP32 Dev Module, compiled in the Arduino IDE 1.8.19  using TFT_eSPI 2.5.0  driving a 480x240 TFT touch panel (ST7796_DRIVER) with SPI interface in use.

I am writing a class for handling the menu screens for my application.   When I called a method that was updating a screen sprite, it would work fine on the first call, but if I called the method again, it would crash the ESP32 (looping endlessly into reset, run the code, reset, run the code...).

I wrote a very simple test case that does nothing but create a globally scoped screen object and sprite object.  Then I pass the sprite to a class method that simply fills the sprite and pushes it.   That's enough to cause the crash.

Note:  In the main test_class.ino, If I comment out the calls to the class method and call the fill/push directly it all works fine.    Alternately,  if I create two sprites and pass each to the class method, it works fine.   The crash only seems to happen when calling the method twice passing the same sprite to it.

Any hints on what might be going on would be greatly appreciated.  Thanks.   

Here are the simple code files:

### library/MyClass/MyClass.h
```
#ifndef MyClass_H
#define MyClass_H

#include "Arduino.h"
#include <TFT_eSPI.h>     // Bodmer Graphics lib

class MyClass {
   public:
      MyClass(const char * title);
      void drawSprite(TFT_eSprite,int,int);
   private:
      const char * _title;
};
#endif
```

### library/MyClass/MyClass.cpp
```
#include "MyClass.h"
MyClass::MyClass(const char * title){
   _title = title;
}
void MyClass::drawSprite(TFT_eSprite sp, int X, int Y) {
   sp.fillSprite(TFT_BLUE);
   sp.pushSprite(X,Y);
}
```

### test_class/test_class.ino
```
#include <TFT_eSPI.h>     // Bodmer Graphics lib
#include <MyClass.h> 
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sp1 = TFT_eSprite(&tft);
TFT_eSprite sp2 = TFT_eSprite(&tft); 

MyClass screen1("Main Menu");

void setup() {
   tft.init();
   tft.setRotation(1);  // Landscape

   sp1.createSprite(100,50);
   sp2.createSprite(100,50);

   tft.fillScreen(TFT_BLACK);

   //sp1.fillSprite(TFT_BLUE);  //<<<  Works fine executed without using the class methods
   //sp1.pushSprite(100,100);   //<<<
   //sp1.pushSprite(200,200);   //<<<

   screen1.drawSprite(sp1,100,100);
   screen1.drawSprite(sp1,200,200);     //<<< Comment this call and replace with 2nd sprite
   //screen1.drawSprite(sp2,200,200);   //<<< and the class methods work...
}

void loop() {
}
```

Bodmer, 2023-04-04:
I don't see this as an issue with the library.

My solution would be to pass instance pointers or inherit from the library. You will need to find a solution that suits your application.

Note that this is intended as an Arduino library rather than a general purpose resource.

closed 2023-04-04 by Bodmer

dfreitas-git, 2023-04-04:
Thank you for your quick reply.   I don’t quite understand what was causing the ESP to reset, but I modified the code to use instance pointers and that worked well.

Thanks again.
    Dan