Bodmer/TFT_eSPI #2575: Unable to initiate st7789 on Wemos D1 mini ESP8266

Mearntain, 2023-04-10:
To minimise effort to resolve issues the following should be provided as a minimum:
1. Can not get tft_espi to work with my esp8266 wemos d1 mini and adafruit 135x240 st7789. Works just fine with adafruit library. 
2. Arduino 2.0.4
3. TFT_eSPI library version 2.5.0
4. Board package version 3.1.2 (esp8266)
5. ESP8266 Wemos D1 Mini
6. TFT driver - st7789
7. Interface type SPI - sclk to D5, mosi to D7, cs to D8, rst to D4, DC to D3 

I just want to run any basic example - display an image

Example output:

```
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_INFO "User_Setup"

// Define to disable all #warnings in library (can be put in User_Setup_Select.h)
//#define DISABLE_ALL_LIBRARY_WARNINGS

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
//#define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
//#define NUCLEO_64_TFT
//#define NUCLEO_144_TFT

// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTB_DATA_BUS

// Tell the library to use parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT
//#defined TFT_PARALLEL_16_BIT // **** 16 bit parallel ONLY for RP2040 processor ****

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
//#define ILI9341_DRIVER       // Generic driver for common displays
//#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
[... 305 lines cut ...]
// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

```
[image]

Bodmer, 2023-04-10:
Try it again after commenting out the following lines in your setup file thus:
//#define TFT_MISO PIN_D6
//#define TFT_MOSI PIN_D7
//#define TFT_SCLK  PIN_D5

Also run Read_User_Setup again and check the reported gpio numbers are correct.

Mearntain, 2023-04-10:
The 3 lines have been commented out, ran Read_User_Setup again, but still am unable to get the display to work with the library.

[image]

Bodmer, 2023-04-10:
I see that the OP states:
TFT_eSPI library version 2.5.0

Yet the Read _User_Setup indicates the version is 2.4.79

Make sure you do not have multiple copies of the library loaded.

I use IDE 1.8.xx but I will try with the new 2.0.4 and see if that makes a difference.

Mearntain, 2023-04-10:
Yep, you are correct that I did have a typo, but I do only have 1 library installed. I downgraded to .79 last night before posting to see if it would fix my issue, and it didnt. .

On another  computer with the same setup.h, I am also having the same issue with Arduino 1.8.xx. That computer never had tft_espi until today so that was a fresh library install there.

For what its worth, I did have this working on the same setup in the past, but I remember there being a "got ya" in the setup file to where I had to add something special but can't remember what it was. 

And if it helps any, if you are familiar with the Adafruit_ST7789 library basic example, If i use option  one with hardware SPI (plugged into same SPI plugs as my setup.h file, the display doesn't work. If I choose option 2 and explicitly define SCLK and MOSI, it works as expected.

Mearntain, 2023-04-10:
This is the display

https://learn.adafruit.com/adafruit-1-14-240x135-color-tft-breakout/pinouts

Here is my specific board

[image]

Bodmer, 2023-04-11:
Ah! You say, "If i use option one with hardware SPI (plugged into same SPI plugs as my setup.h file, the display doesn't work. If I choose option 2 and explicitly define SCLK and MOSI, it works as expected."

The second option works because the Adafruit library will then use software SPI which can be used on any pins.  The fact that the first option does not work shows that the hardware SPI pins are not correctly wired to the display.

How can this be? The answer I think is that the Wemos D1 mini comes indifferent versions. Older boards mapped different pins to say D7!

Check you Wemos board version and try to find the GPIO pin number map to your PCB, I think the pin map posted above is incorrect for the board you have. Then use the GPIO numbers ( not Dx or PIN_Dx references) in the setup file.

Mearntain, 2023-04-11:
That is the pinout that was given to me when I originally purchased the chips. Any idea of how to tell which one this is? I've tried every combination of pinouts I could find online and yet to make any work. 
[image]
[image]

Mearntain, 2023-04-11:
UPDATE  - I  got home, and using a raw chip pinout for the wifi chip, I beep tested MOSI to truly be D7 as expected along with SCLK beeping  out to D5 as expected. I have even substituted  actualy GPIO pin numbers in the setup as well and still get  the same result. Even more stumped now..

Mearntain, 2023-04-12:
UPDATE AGAIN - using a multimeter, I am getting what I assume to be proper readings from my MOSI (D7  - GPIO 13) and SCLK (D5 - GPIO 14) pins (all pins defined read about 1.6V while all non-defined pins measure the full 5V). This really makes me think this is a issue with the setup.h file

Bodmer, 2023-04-12:
A logic 1 will be about 3.3V so measuring from 5V to a logic 1 would give about 1.7V which is what you measure.

There are changes in the latest ESP8266 board package. The board architecture is now signalled by:
```
#define ARDUINO_ARCH_ESP8266
```
It used to be:
```
#define ESP8266
```
It looks like the Adafruit library still uses lines like:
```
#if defined (ESP8266)
  ...
#endif
```
So that means Adafruit_GFX is not completely compatible with the latest ESP8266 board package. This would explain why hardware SPI does not work.

The latest github master library has been updated to handle old and new board packages.

Maybe try older board packages?

Bodmer, 2023-04-12:
I'll connect up a ESP8266 board to an ST7789 display and see if it works for me.

Which board do you select in the menu:
[image]

Mearntain, 2023-04-12:
I have tried several boards in the menu all with the same result, but LOLIN(WEMOS) D1 mini (clone) is my normal go-to for uploading

Bodmer, 2023-04-12:
I have connected an ST7789 screen to an ESP8266 and it is working fine.

Sequence:
1. Use Arduino IDE 1.8.13
2. Download fresh copy of TFT_eSPI v 2.5.0 using library manager
3. Download frech copy of ESP8266 board package 3.1.2
4. Make the following chnges to User_Setup.h:
   a) Comment out ILI9341 driver at line 45
   b) Uncomment ST7789 driver at line 55
5. Connect display to an ESP8266 board (NodeMCU format but same GPIO used).
6. Select ESP8266 board package
7. Select LOLIN(WEMOS) D1 mini (clone) board
8. Set upload speed to 921600 (just to speed up loading) and COM9
9. Upload "Sprite_Draw" sketch
10.  Display runs fine but with blue and red swapped
11.  Make the following chnges to User_Setup.h:
      a) Uncomment line 77 to select BGR colour order
12. Upload "Sprite_Draw" sketch and display works with correct colours

My display has no 5V to 3.3V regulator so it is powered from 3.3V only.

Next change to Arduino IDE 2.0.3 (already configured to use the same sketch/libraryfolder and board packages as the older IDE).
1. Select ESP8266 board package
2. Set upload speed to 921600 (just to speed up loading) and COM9
3. Upload "Colour_Wheel" sketch and display runs fine

I have a 240x240 display but did NOT set the size in the setup file so it will be treated as 240x320 and not all drawn graphics are visible as expected.

So it appears I have wired up the display the same as yours and it works fine.

[image]

[image]

The only difference is that I used a NodeMCU board as I do not have any Lolin boards. However I would not expect that to make a difference.

Do check your display has a 5v to 3.3V regulator if you are powering from 5V. Make sure the display backlight does not flicker during program execution as that would indicate an inadequate power supply.

Other than that I don't think I can help further.

Mearntain, 2023-04-12:
Sounds good - my  display does have a regulator on it. Even powering with 3.3 instead,  no  change. Tried again with  all of your mentioned settings and no new  results  (see screenshot, was  unable to  include tft_espi in same screenshot but  that is at  2.5.0 fresh install).

[image]

I wish I could chalk it up to bad hardware or something like that,  but unfortunately that is not that the case as it works perfectly fine (just compiled this after the failed tft_espi attempt) with a different libary. And the fact that I did use this library about a year ago with the exact same hardware,  I know it is possiible. I remember searching Google for hours on end and founds tons of people with similar issues, but all  seemed to have the same fix. I was only able to find one post  (or comment maybe) that had a direct reesolution to my issue. I really wish I  could find that link and have been searching over a week now. Only  hope is maybe my computer still has browser history saved from that far back to find it again. 

[image]

Mearntain, 2023-04-12:
I  do recall it being a adafruit brand display issue specific to your library if that rings a bell at all? Maybe I get lucky and find the year old history in the meanwhile

Bodmer, 2023-04-13:
It is supsicious that it does not work with an ESP32 either.

I think the Adafruit displays are 5V tolerant and have inline buffers. Maybe drop the SPI clock speed down to a very low value of 1MHz and see if it works. It could be there is a large delay in the buffers if they are powered froma low voltage?

Bodmer, 2023-04-13:
The display does have a HC4050 level converter so that will limit the maximum SPI rate as I expect it is run off 3.3V.

Mearntain, 2023-04-13:
Well.....I made progress....but unsure of what it was the caused that progress....BUT Sprite example does work now.

I'm making an assumption that there is some sort of time delay issue (too short) or something causing the problem. Or maybe I'm completely wrong?

Upload...nothing happened. Reset button, nothing happens. (THIS NEXT PART MIGHT BE IMPORTANT) Unplugging power and then plugging back  in, the LED on board stays solid blue as if boot failed or froze maybe? I hit the reset button, blue light returns to normal operation but display still doesnt work. Wait a few more seconds, reset button, and Sprite starts displaying. At this point I can hit reset a couple more times and the sketch resets and runs as expected and I can repeat as many times as I want. 

[image]

Bodmer, 2023-04-13:
Since the Read_User_Setup sketch runs as reported above this shows boot occurs correctly after sketch upload. 

Did you try a much lower SPI clock speed?

Another option is to change this line:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L132
To invoke mode 0 rather than 3.

Mearntain, 2023-04-13:
Yep, I did try a much lower SPI speed, down to 1mhz as you suggested. That was when I was able to get the display to work from the previous picture.

I changed SPI mode to 0 and that caused the display to power up immediately after pressing the reset button. I played with multiple speeds afterwards and 1mhz seems to be where its at (faster wasn't displaying). I do still have to press reset twice to make it power up, but not with the long delays between now after changing SPI mode. I supposed maybe trying a different reset pin may fix that.

At least its working  now and I'll play with the settings to get it optimized. I appreciate you hanging in there with me!

Bodmer, 2023-04-13:
Re. SPI mode. You can invoke mode 0 by adding:
```
#define TFT_SPI_MODE SPI_MODE0
```
To the setup file.

Software SPI with the Adafruit library would probably operate at a few MHz.

I suspect the HC4050 logic level translator is introducing signal skew. Powering the display from 5V will raise the chip supply voltage slightly and then the buffer delay will reduce slightly.

You may find changing the GPIO for one of the pins DC, CS or RST solves the boot problem. This is because some of the GPIO have alternative functions at boot time and are required to be in a certain state immediately after power on. This seems to fit with your boot up issues.

 You can buy ST7789 displays that do not have the buffer cheaply from China if you are patient with the long postal delay. I have also seen them on Amazon. Purchasing and integrated display plus processor is also relatively inexpensive.

I am going to close this as I think this is a hardware level problem. I think the ESP8266 can bitbash SPI at around 6MHz, so try that speed with hardware SPI. If you solve it or have more info on the required operating constraints then do share.

closed 2023-04-13 by Bodmer

Mearntain, 2023-04-13:
I did a little playing around just now and it appears that the board does work at 6mhz as you suspected. That seems to be the cutoff and I am unable to get it to display past that. Thanks again for  the help in getting things  going!