Bodmer/TFT_eSPI #2585: Wemos D1 R32 with 3.5" TFT LCD Shield

vgomasjaya, 2023-04-15:
I have wemos d1 r32 and 3.5" TFT LCD Shield... 
After I run Read_User_Setup, I got this

`[code]
TFT_eSPI ver = 2.5.0
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 9341
Display width  = 240
Display height = 320

TFT_CS   = GPIO 33
TFT_DC   = GPIO 15
TFT_RST  = GPIO 32
TFT_WR   = GPIO 4
TFT_RD   = GPIO 2

TFT_D0   = GPIO 12
TFT_D1   = GPIO 13
TFT_D2   = GPIO 26
TFT_D3   = GPIO 25
TFT_D4   = GPIO 17
TFT_D5   = GPIO 16
TFT_D6   = GPIO 27
TFT_D7   = GPIO 14

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled`

At first I think my LCD driver is ILI9486, but after I get this output, I realize that it's 9341... I already wiring my Wemos D1 R32
- IO15 wired to IO35
- IO33 wired to IO34
- IO32 wired to IO36
- IO4 wired to IO39 (for touchscreen)

And in User_Setup_Select.h I already uncomment one by one
- #include <User_Setups/Setup13_ILI9481_Parallel.h>  // Setup file for the ESP32 with parallel bus TFT
- #include <User_Setups/Setup14_ILI9341_Parallel.h>  // Setup file for the ESP32 with parallel bus TFT

But my LCD just get white background and nothing happen even I run Colour_test

Please help me and thank you before

Bodmer, 2023-04-15:
It looks like you have not commented out [this line 30](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L30).

As noted in line 28: "Only ONE line below should be uncommented"

vgomasjaya, 2023-04-16:
I've uncommented it and still get white screen... but when I use mcufriend_kbv, it works without any mod... but now the problem is the touchscreen doesn't work properly

Bodmer, 2023-04-19:
I asked toI comment out [this line 30](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L30).

NOT uncomment it!

It should look like this:
//#include <User_Setup.h>           // Default setup is root library folder

To be clear on what you have done, post your complete User_Setup_Select.h file and the setup file you want to use as ZIP files. (Drag and drop).

Bodmer, 2023-04-25:
Assumed resolved.

closed 2023-04-25 by Bodmer