Bodmer/TFT_eSPI #2589: Pixelation along two edges of 7735 display

phillmj, 2023-04-16:
Hi, fantastic lib as I'm always telling you!

I'm trying a new 7735 screen found here: [https://wiki.dfrobot.com/0.96_Inch_160_80_Color_SPI_TFT_Display_SKU_DFR0847#target_6](url)

I have tried all the possible definitions of the 7735:

```
/ #define ST7735_INITB
// #define ST7735_GREENTAB
// #define ST7735_GREENTAB2
// #define ST7735_GREENTAB3
// #define ST7735_GREENTAB128    // For 128 x 128 display
#define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
// #define ST7735_ROBOTLCD       // For some RobotLCD arduino shields (128x160, BGR, https://docs.arduino.cc/retired/getting-started-guides/TFT)
// #define ST7735_REDTAB
// #define ST7735_BLACKTAB
// #define ST7735_REDTAB160x80   // For 160 x 80 display with 24 pixel offset
```

All of them exhibit this streaking (one doesn't work at all, and some actually have the screen offset by a lot on two sides). One I'm using is #define ST7735_GREENTAB160x80 

When the display first powers up, the backlight comes on and there's no streaking, so I don't believe it's a faulty display. Also, for some of the above defines, the streaking is different.

Any ideas how to get rid of the streaking? I can't seem to find a way to fix this.

Thanks for your tireless support and amazing libraries. I'm a fan for life.

Mike

[image]

PaulVdBo, 2023-04-16:
Same problem here... but on opposite sides...

phillmj, 2023-04-16:
I fixed it.

In the Init and Rotation files, wherever I found a reference to INITR_GREENTAB160x80  with colstart and rowstart, I changed it like this:
```
else if (tabcolor == INITR_GREENTAB160x80)
       {
         commandList(Rcmd2green);
         writecommand(TFT_INVON);
         **colstart = 24;** //26
         **rowstart = 0;** //1
```