Bodmer/TFT_eSPI #2618: Random block on text

Keaster20, 2023-04-30:
Using a 4.0" tft with ILI9488 chip I get this random block on the last letter that is written to the screen. Does not matter is I use print or drawstring. I have never had this problem with other displays such as ili9341. If I display more text the block moves to the last character displayed.
[image]

Bodmer, 2023-04-30:
Provide an exampel sketch that demonstrates the problem. Other information that is helpful to track down the reason is listed here:
https://github.com/Bodmer/TFT_eSPI/blob/master/.github/ISSUE_TEMPLATE/issue-template.md

Bodmer, 2023-05-02:
Has this been solved?

Bodmer, 2023-05-11:
Assumed resolved.

closed 2023-05-11 by Bodmer