Bodmer/TFT_eSPI #2620: IL9143 & Touch & SD wont run

sundeepgoel72, 2023-05-01:
Hi,

I have the same issue as #762. 

My setup is 
1. esp32 WROOM devkit
2. Tft 2.8

connections are as follows
- TFT and Touch share same SPI bus, different CS
- SD card is on a different bus with different CS

TFT and Touch work just fine, till SD.begin is called. After that the touch freezes but the TFT keeps working fine.

I tried various tips from the net e.g. driving CS high explicitly, however, nothing works. I then hooked by the board to an oscilloscope. The touch CS is being driven , however, there seems to be change in the pattern

When touch not working
[image]

When touch working
[image]

Any pointers on how to resolve the issue ?

sundeepgoel72, 2023-05-01:
Please ignore - i read the user_setup carefully and discovered the following 
```
// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.
// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect
```
after uncommenting following, everything is working fine
`//#define SUPPORT_TRANSACTIONS`

Q:  per above for ESP32 are automatically enabled, however, that did not seem to be happening. or maybe I have missed some config parameter. Please also elaborate - what happens when transactions are enabled ?

Bodmer, 2023-05-01:
The issue symptoms are rather odd.

SUPPORT_TRANSACTIONS will be defined by the library automatically by the library for the ESP32, so it does not matter if it is not defined in the setup file. Also the touch functionality built into the library does not need transaction support to be enabled.

Just because the SD card is on a different set of pins does not mean that it is using a different SPI hardware block. Perhaps this is the problem. There are 2 SPI hardware blocks in the ESP32, viz VSPI and HSPI.  Libraries (including TFT_eSPI) typically to use the VSPI hardware by default, so maybe that is the root cause of the problem?

Is the TFT configured to use the standard VSPI bus hardware or have you forced it to use the HSPI bus hardware in the setup file?

Bodmer, 2023-05-01:
FYI this is where it is ensured that SUPPORT_TRANSACTIONS is defined:

https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L36-L39

sundeepgoel72, 2023-05-01:
> Is the TFT configured to use the standard VSPI bus hardware or have you forced it to use the HSPI bus hardware in the setup file?

Hi, thanks for the prompt reply. 

I had enabled USE_HSPI_PORT as well in the config. I did a test after commenting it out, and still retaining SUPPORT_TRANSACTIONS - **touch stopped working.** 

I then commented out SUPPORT_TRANSACTIONS, and enabled USE_HSPI_PORT . **Touch worked properly again.** 

So as you suspected, **the issue is resolved by enabling USE_HSPI_PORT.** Could you please explain what this does at a code / hardware level ?

Bodmer, 2023-05-01:
In summary:
There are multiple hardware SPI blocks in the ESP32, these can be mapped to drive chosen GPIO. If two libraries try to map a SPI block to different pins then there is a resource conflict.

Google for a more detailed explanation.

closed 2023-05-01 by Bodmer