Bodmer/TFT_eSPI #2621: New User Setup creates 'VSPIQ_IN_IDX not declared' error

MisterDiz, 2023-05-02:
I'm trying to create a User_Setup for a JCZN 1.9" TFT running on an ESP32 S3. Its similar in looks to a LilyGo T-Display S3 but SPI, rather than parallel and without battery connection and seemingly unconventional pin assignment.

[image]

The core of my setup file looks this:

```
#define USER_SETUP_ID 902 // JCZN 1.9"

#define ST7789_DRIVER
#define CGRAM_OFFSET // Library will add offsets required
#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red
#define TFT_INVERSION_ON

#define TFT_WIDTH 170
#define TFT_HEIGHT 320

#define TFT_MOSI 13
#define TFT_SCLK 12
#define TFT_CS 10
#define TFT_DC 11 
#define TFT_RST 1 

#define TFT_BL 14 
#define TFT_BACKLIGHT_ON HIGH
```

The vendor provides an example which works with the Arduino GFX library but when I try and compile with TFT_eSPI in Arduino IDE 2.1.0, I get multiple errors of this form:

```
In file included from /Users/xxxx/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:20:
/Users/xxxx/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'void TFT_eSPI::begin_SDA_Read()':
/Users/xxxx/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:67:31: error: 'VSPIQ_IN_IDX' was not declared in this scope
   pinMatrixInAttach(TFT_MOSI, VSPIQ_IN_IDX, false);
                               ^~~~~~~~~~~~
```

Could you please point me in the right direction of what I am doing wrong? TIA

Bodmer, 2023-05-02:
I suspect this is down to the combination of ESP32 board package version in use and the TFT_eSPI library version.

See here for latest line:
https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32_S3.c#L66

Also that error line should not be seen by the compiler [unless TFT_SDA_READ is defined](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32_S3.c#L56). So this indicates multiple setup files are being invoked somewhere.

MisterDiz, 2023-05-03:
Thanks Bodmer, nailed it. All working fine now. All your efforts with this amazing library are much appreciated 🥇

closed 2023-05-03 by MisterDiz