Bodmer/TFT_eSPI #2624: Kedei 3.5 TFT not working !!

sundeepgoel72, 2023-05-03:
Hi,

I have an issue similar to #610 

The display is a keDei rpi TFT display with touch.

[image]

Per internet search this display should be using ili9846, am using a ESP32 WROOM devkit board.

config used is 
```
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    27  // Dummy since board does not have this pin exposed.
#define TFT_RST   26  // Dummy since board does not have this pin exposed.
#define TOUCH_CS 33     // Chip select pin (T_CS) of touch screen
```
Read somewhere on the internet this display uses 3 wire spi vs 4 wire  - dont know what that means since MOSI / MISO and CS are present, however DC and RST are not.

getting blank display , any suggestions for troubleshooting.

sundeepgoel72, 2023-05-03:
update : I tried following drivers
RPI_ILI9486 - TFT not working , touch working 
ILI9481 - TFT not working , touch working. However, TFT does flicker, indicating something was "happening"
ILI 9486 - TFT not working , touch working 
ILI 9488 - TFT not working , touch working

sundeepgoel72, 2023-05-03:
same issue with another rpi display :-(

[image]

at least this one flickers, however, keeps reseting

```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

Bodmer, 2023-05-03:
The Kedei display is not compliant with the Waveshare design so will not work. See ReadMe for more information.

The second one should work. The reset likely has another root cause, maybe insufficient power, sketch problem etc.

sundeepgoel72, 2023-05-04:
> The Kedei display is not compliant with the Waveshare design so will not work. See ReadMe for more information.

Thanks - assume this section is the relevant one 
_The ILI9486 RPi display must be of the Waveshare design and use a 16 bit serial interface based on the 74HC04, 74HC4040 and 2 x 74HC4094 logic chips. Note that due to design variations between these displays not all RPi displays will work with this library, so purchasing a RPi display of these types solely for use with this library is NOT recommended._

Do you think there is any merit in me trying to create a custom driver for this, or is the design way to different for this to be practical. 
if there is merit - could you give me some pointers to get started?

sundeepgoel72, 2023-05-04:
> The second one should work. The reset likely has another root cause, maybe insufficient power, sketch problem etc.

Following are the pin connections am currently using
[image]

I'll give this another shot...

sundeepgoel72, 2023-05-04:
> I'll give this another shot...

Looked at all connections, voltage levels and sketch. issue persists, continuous Resets and screen flickering.

Bodmer, 2023-05-04:
Post pictures of the wiring. Some displays need 3.3V and 5V connected. I am not sure if you have one of those.

Disconnect the screen and run the same sketch (which one?) does it still crash?

Post you complete setup file.

Detail your software environment, info guidelines here:
https://github.com/Bodmer/TFT_eSPI/blob/master/.github/ISSUE_TEMPLATE/issue-template.md

Bodmer, 2023-05-14:
Assumed resolved.

closed 2023-05-14 by Bodmer

Luma48, 2025-05-21:
Heya I know this is random but did you ever manage to get the Kedei 3.5 TFT LCD working. Now I'm pretty sure my wiring is correct here:
| LCD Pin | Meaning           | ESP32 GPIO |
| ------- | ----------------- | ------------------ |
| L_CS    | LCD Chip Select   | GPIO 15            |
| T_CS    | Touch Chip Select | GPIO 4             |
| IRQ     | Touch Interrupt   | GPIO 21            |
| GND     | Ground            | GND                |
| CLK     | SPI Clock         | GPIO 5 (SCK)       |
| MISO    | SPI Master In     | GPIO 19 (MISO)     |
| MOSI    | SPI Master Out    | GPIO 23 (MOSI)     |
| 5V      | Power             | 5V pin             |
| NC      | Not Connected     | Left unconnected  |

And when run my program I wrote and the screen just shows white (aka I'm pretty sure that's the back light). This is what my code for my ESP32 looks like when everything is all connected:

#include <TFT_eSPI.h>              // TFT library (configured via custom setup)
#include <XPT2046_Touchscreen.h>    // Touch library
#include <SPI.h>

// Initialize display and touch
TFT_eSPI tft = TFT_eSPI();         // Initialize TFT library
XPT2046_Touchscreen ts(TOUCH_CS, TOUCH_IRQ);  // Initialize touch screen

void setup() {
  Serial.begin(115200);            // Start serial monitor at 115200 baud
  delay(1000);

  // Initialize the TFT display
  tft.init();                      // Begin communication with the display
  tft.setRotation(1);              // Set display to landscape mode
  tft.fillScreen(TFT_BLACK);       // Clear screen with black
  tft.setTextColor(TFT_WHITE, TFT_BLACK);  // White text on black background
  tft.setTextSize(2);              // Set text size
  tft.drawString("Touch the screen...", 50, 10);  // Display a prompt on the screen

  // Initialize the touch screen
  ts.begin();                      // Begin communication with XPT2046
  ts.setRotation(1);               // Set touch screen rotation
}

void loop() {
  if (ts.touched()) {
    // Only proceed if the screen is touched
    TS_Point p = ts.getPoint();  // Get the touch coordinates

    // Map raw touch coordinates to screen coordinates
    int x = map(p.x, 200, 3900, 0, 480);  // Adjust calibration range as needed
    int y = map(p.y, 200, 3900, 0, 320);

    // Bound values to screen size
    x = constrain(x, 0, 479);
    y = constrain(y, 0, 319);

    // Clear screen area for coordinates display
    tft.fillRect(0, 40, 480, 40, TFT_BLACK);
    
    // Show coordinates on the TFT display
    tft.setCursor(20, 50);
    tft.printf("Touch at: X=%d, Y=%d", x, y);

    // Draw a dot at the touch location
    tft.fillCircle(x, y, 3, TFT_RED);

    // Print to Serial Monitor only when touched
    Serial.print("Touch: x=");  // Print "Touch: x="
    Serial.print(x);            // Print the X coordinate
    Serial.print(", y=");       // Print ", y="
    Serial.println(y);          // Print the Y coordinate and add a newline

    delay(100);  // Small delay to debounce
[... 37 lines cut ...]
// === TOUCHSCREEN PINS (XPT2046) ===
#define TOUCH_CS  4               // T_CS -> GPIO 4  (Touch chip select)
#define TOUCH_IRQ 21              // IRQ  -> GPIO 21 (Touch interrupt, optional but used here)

// === Display Size ===
//#define TFT_WIDTH  480   // Set the screen width (replace with your display's width)
//#define TFT_HEIGHT 320   // Set the screen height (replace with your display's height)

// === SPI FREQUENCIES ===
#define SPI_FREQUENCY       27000000    // SPI speed for the TFT
#define SPI_TOUCH_FREQUENCY 2500000     // SPI speed for the touch controller

Now if I'm right what should be happening here is the pins are properly set and I believe I am using the correct driver (that being ILI9486) and the screen should be black with some white text saying to touch it (since this is a touch screen), the loop just checks to see if the screen is touched and draws a red dot in the point it is.

All this aside why I'm really here is for some guidance since your screen looks pretty much the same as mine and yet nothing displays on mine??? I wanted to know if perhaps you have the Datasheet, maybe have written some custom libs and or drivers for it to work properly with an ESP32. If not maybe you still got it working using TFT_eSPI.h and I'd be curious to learn how. I'll leave some diagrams of wiring and the board itself so you can maybe get a better understanding of what I'm working with. Thanks again if you read this!!!

I have an ESP32 WROOM 32 here are some details:
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: Redacted

[image]

[image]