Bodmer/TFT_eSPI #2628: ESP-IDF gives the error "GPIO was not declared in this scope" when compiling with ESP-IDF v5.1

devrim-oguz, 2023-05-04:
Hello, thank you for creating this library.

I've been using this library with ESP-IDF v4.4 in a project for a while. But when I upgrade to the ESP-IDF version v5.1, the compilation gives the following errors:

```
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:446:20: error: 'GPIO' was not declared in this scope
  446 |       #define RD_L GPIO.out_w1tc = (1 << TFT_RD)
      |                    ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:98:3: note: in expansion of macro 'RD_L'
   98 |   RD_L;
      |   ^~~~
In file included from /home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp:20:
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:100:9: error: 'gpio_input_get' was not declared in this scope; did you mean 'gpio_num_t'?
  100 |   reg = gpio_input_get(); // Read three times to allow for bus access time
      |         ^~~~~~~~~~~~~~
      |         gpio_num_t
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'void TFT_eSPI::pushBlock(uint16_t, uint32_t)':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:412:31: error: 'GPIO' was not declared in this scope
  412 |       #define tft_Write_16(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:547:5: note: in expansion of macro 'tft_Write_16'
  547 |     tft_Write_16(color);
      |     ^~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:412:31: error: 'GPIO' was not declared in this scope
  412 |       #define tft_Write_16(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:558:23: note: in expansion of macro 'tft_Write_16'
  558 |   else while (len--) {tft_Write_16(color);}
      |                       ^~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'void TFT_eSPI::pushSwapBytePixels(const void*, uint32_t)':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:412:31: error: 'GPIO' was not declared in this scope
  412 |       #define tft_Write_16(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:568:20: note: in expansion of macro 'tft_Write_16'
  568 |   while ( len-- ) {tft_Write_16(*data); data++;}
      |                    ^~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'void TFT_eSPI::pushPixels(const void*, uint32_t)':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:412:31: error: 'GPIO' was not declared in this scope
  412 |       #define tft_Write_16(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:578:37: note: in expansion of macro 'tft_Write_16'
  578 |   if(_swapBytes) { while ( len-- ) {tft_Write_16(*data); data++; } }
      |                                     ^~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:416:32: error: 'GPIO' was not declared in this scope
  416 |       #define tft_Write_16S(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 0)); WR_H; \
      |                                ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:579:27: note: in expansion of macro 'tft_Write_16S'
  579 |   else { while ( len-- ) {tft_Write_16S(*data); data++;} }
      |                           ^~~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::spiwrite(uint8_t)':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:392:27: error: 'GPIO' was not declared in this scope
  392 |   #define tft_Write_8(C)  GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t)(C)); WR_H
      |                           ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp:910:3: note: in expansion of macro 'tft_Write_8'
  910 |   tft_Write_8(c);
      |   ^~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::writecommand(uint8_t)':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:167:20: error: 'GPIO' was not declared in this scope
  167 |       #define DC_C GPIO.out_w1tc = (1 << TFT_DC)
      |                    ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp:924:3: note: in expansion of macro 'DC_C'
  924 |   DC_C;
      |   ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::writedata(uint8_t)':
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:168:20: error: 'GPIO' was not declared in this scope
  168 |       #define DC_D GPIO.out_w1ts = (1 << TFT_DC)
      |                    ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp:972:3: note: in expansion of macro 'DC_D'
  972 |   DC_D;        // Play safe, but should already be in data mode
[... 74 lines cut ...]
 4742 |             else {tft_Write_16(textbgcolor);}
      |                   ^~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:412:31: error: 'GPIO' was not declared in this scope
  412 |       #define tft_Write_16(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp:4747:18: note: in expansion of macro 'tft_Write_16'
 4747 |         if (pX) {tft_Write_16(textbgcolor);}
      |                  ^~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:412:31: error: 'GPIO' was not declared in this scope
  412 |       #define tft_Write_16(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp:4796:30: note: in expansion of macro 'tft_Write_16'
 4796 |               while (tnp--) {tft_Write_16(textcolor);}
      |                              ^~~~~~~~~~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:412:31: error: 'GPIO' was not declared in this scope
  412 |       #define tft_Write_16(C) GPIO.out_w1tc = GPIO_OUT_CLR_MASK; GPIO.out_w1ts = set_mask((uint8_t) ((C) >> 8)); WR_H; \
      |                               ^~~~
/home/ado/Projects/Espressif/OPIS_Cloud_Module/main/lib/TFT_eSPI/TFT_eSPI.cpp:4798:19: note: in expansion of macro 'tft_Write_16'
 4798 |             else {tft_Write_16(textcolor);}
      |                   ^~~~~~~~~~~~
```

It seems like, in the new ESP-IDF, the GPIO construct is not readily available to all users like it used to be. And there is a related issue on the idf page: https://github.com/espressif/esp-idf/issues/9184

From what I understand from that issue, the GPIO was never supposed to be available to users. Instead, it is a low-level library. I guess it is used in the library for fast pin updates by using the pin registers directly. But with the new version, it causes an error. I don't know what to do, so I'm writing an issue here. Maybe we can also collaborate with the idf developers to resolve this problem. I'm available if you need to ask anything more. Thanks for your help.

Bodmer, 2023-05-10:
Thanks for raising this. The Espressif software team like to change things that are not broken!

The fix is in the link you provided. I will try this next week.

Bodmer, 2023-05-18:
Adding the include to the relevant processor header files does not break the library for the Arduino environment so I will add it in the next update.

For example for the ESP32 add the following line [here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L14):
```
#include "hal/gpio_ll.h"
```

alba-ado, 2023-05-25:
Hello, I have added the line as you said in the ESP32_S3.h file (my board is esp32-s3) and now the other errors are disappeared but this error came up:

```
In file included from /home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.h:94,
                 from /home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.cpp:16:
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:11:3: warning: #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update) [-Wcpp]
   11 |  #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update)
      |   ^~~~~~~
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.h:909:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  909 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
In file included from /home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.cpp:20:
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:100:9: error: 'gpio_input_get' was not declared in this scope; did you mean 'gpio_num_t'?
  100 |   reg = gpio_input_get(); // Read three times to allow for bus access time
      |         ^~~~~~~~~~~~~~
      |         gpio_num_t
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ado/Projects/Espressif/idf_deneme/build/log/idf_py_stderr_output_24645 and /home/ado/Projects/Espressif/idf_deneme/build/log/idf_py_stdout_output_24645
```

devrim-oguz, 2023-06-19:
Hello, I have added the line as you said in the ESP32_S3.h file (my board is esp32-s3) and now the other errors are disappeared but this error came up:

```
In file included from /home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.h:94,
                 from /home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.cpp:16:
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:11:3: warning: #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update) [-Wcpp]
   11 |  #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update)
      |   ^~~~~~~
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.h:909:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  909 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
In file included from /home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/TFT_eSPI.cpp:20:
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
/home/ado/Projects/Espressif/idf_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:100:9: error: 'gpio_input_get' was not declared in this scope; did you mean 'gpio_num_t'?
  100 |   reg = gpio_input_get(); // Read three times to allow for bus access time
      |         ^~~~~~~~~~~~~~
      |         gpio_num_t
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ado/Projects/Espressif/idf_deneme/build/log/idf_py_stderr_output_24645 and /home/ado/Projects/Espressif/idf_deneme/build/log/idf_py_stdout_output_24645
```

alba-ado, 2023-07-05:
I'm devrim-oguz, this is my business account. Here is an update of the errors when I switched to v5.1-rc2:

```
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/TFT_eSPI.h:97,
                 from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/TFT_eSPI.cpp:16:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:111:3: warning: #warning >>>>------>> DMA is not supported in parallel mode [-Wcpp]
  111 |  #warning >>>>------>> DMA is not supported in parallel mode
      |   ^~~~~~~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/TFT_eSPI.h:977:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  977 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/TFT_eSPI.cpp:20:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:96:8: error: 'gpio_get_level' was not declared in this scope; did you mean 'gpio_ll_get_level'?
   96 |   b  = gpio_get_level((gpio_num_t)TFT_D0); // Read three times to allow for bus access time
      |        ^~~~~~~~~~~~~~
      |        gpio_ll_get_level
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stderr_output_18292 and /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stdout_output_18292
```

It says to change the gpio_get_level commands to the ones with ll, so I did. But now it gives this error, and I don't know how to convert the pin numbers into (gpio_dev_t *)

```
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/TFT_eSPI.cpp:20:
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:96:26: error: cannot convert 'gpio_num_t' to 'gpio_dev_t*' {aka 'volatile gpio_dev_s*'}
   96 |   b  = gpio_ll_get_level((gpio_num_t)TFT_D0); // Read three times to allow for bus access time
In file included from /home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:16:
/home/ado/Programs/Espressif/esp-idf-v5.1/components/hal/esp32s3/include/hal/gpio_ll.h:337:49: note:   initializing argument 1 of 'int gpio_ll_get_level(gpio_dev_t*, uint32_t)'
  337 | static inline int gpio_ll_get_level(gpio_dev_t *hw, uint32_t gpio_num)
      |                                     ~~~~~~~~~~~~^~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:97:26: error: cannot convert 'gpio_num_t' to 'gpio_dev_t*' {aka 'volatile gpio_dev_s*'}
   97 |   b  = gpio_ll_get_level((gpio_num_t)TFT_D0);
/home/ado/Programs/Espressif/esp-idf-v5.1/components/hal/esp32s3/include/hal/gpio_ll.h:337:49: note:   initializing argument 1 of 'int gpio_ll_get_level(gpio_dev_t*, uint32_t)'
  337 | static inline int gpio_ll_get_level(gpio_dev_t *hw, uint32_t gpio_num)
      |                                     ~~~~~~~~~~~~^~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:98:26: error: cannot convert 'gpio_num_t' to 'gpio_dev_t*' {aka 'volatile gpio_dev_s*'}
   98 |   b  = gpio_ll_get_level((gpio_num_t)TFT_D0); // Data should be stable now
/home/ado/Programs/Espressif/esp-idf-v5.1/components/hal/esp32s3/include/hal/gpio_ll.h:337:49: note:   initializing argument 1 of 'int gpio_ll_get_level(gpio_dev_t*, uint32_t)'
  337 | static inline int gpio_ll_get_level(gpio_dev_t *hw, uint32_t gpio_num)
      |                                     ~~~~~~~~~~~~^~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:101:27: error: cannot convert 'gpio_num_t' to 'gpio_dev_t*' {aka 'volatile gpio_dev_s*'}
  101 |   b  = (gpio_ll_get_level((gpio_num_t)TFT_D0) << 0);
/home/ado/Programs/Espressif/esp-idf-v5.1/components/hal/esp32s3/include/hal/gpio_ll.h:337:49: note:   initializing argument 1 of 'int gpio_ll_get_level(gpio_dev_t*, uint32_t)'
  337 | static inline int gpio_ll_get_level(gpio_dev_t *hw, uint32_t gpio_num)
      |                                     ~~~~~~~~~~~~^~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:102:27: error: cannot convert 'gpio_num_t' to 'gpio_dev_t*' {aka 'volatile gpio_dev_s*'}
  102 |   b |= (gpio_ll_get_level((gpio_num_t)TFT_D1) << 1);
/home/ado/Programs/Espressif/esp-idf-v5.1/components/hal/esp32s3/include/hal/gpio_ll.h:337:49: note:   initializing argument 1 of 'int gpio_ll_get_level(gpio_dev_t*, uint32_t)'
  337 | static inline int gpio_ll_get_level(gpio_dev_t *hw, uint32_t gpio_num)
      |                                     ~~~~~~~~~~~~^~
/home/ado/Projects/Espressif/idf-v5.1_deneme/main/lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:103:27: error: cannot convert 'gpio_num_t' to 'gpio_dev_t*' {aka 'volatile gpio_dev_s*'}
  103 |   b |= (gpio_ll_get_level((gpio_num_t)TFT_D2) << 2);
[... cut ...]
/home/ado/Programs/Espressif/esp-idf-v5.1/components/hal/esp32s3/include/hal/gpio_ll.h:337:49: note:   initializing argument 1 of 'int gpio_ll_get_level(gpio_dev_t*, uint32_t)'
  337 | static inline int gpio_ll_get_level(gpio_dev_t *hw, uint32_t gpio_num)
      |                                     ~~~~~~~~~~~~^~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stderr_output_18384 and /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stdout_output_18384
```

alba-ado, 2023-07-05:
Another update: It gives the same error when I call SPIFFS.begin(); without including this library. So, I think the error comes from the Arduino core for the esp32. These two fixes should fix the library but I can't test it now. So I am not creating a pull request, but maybe you might add these lines in the future.

pablobatch, 2023-10-16:
> Adding the include to the relevant processor header files does not break the library for the Arduino environment so I will add it in the next update.
>
> For example for the ESP32 add the following [...]

Doing this the error dissapeares, but the display of my Lilgo T-Display is not working.

tonhuisman, 2023-11-16:
> Another update: It gives the same error when I call SPIFFS.begin(); without including this library.

SPIFFS is removed from IDF 5.x. It has been deprecated for some years, and is now (finally/actually) removed. You should switch to LittleFS, but there is no migration path available (AFAICS).

devrim-oguz, 2023-11-16:
> > Another update: It gives the same error when I call SPIFFS.begin(); without including this library.
> 
> SPIFFS is removed from IDF 5.x. It has been deprecated for some years, and is now (finally/actually) removed. You should switch to LittleFS, but there is no migration path available (AFAICS).

Not me, but the library needs to switch. It is using the SPIFFS internally. So, it wouldn’t compile for IDF v5.1

kdg3737, 2023-12-13:
I can confirm that I got this library working with the LilyGO T-display-S3 screen on ESP-IDF v5.1, ran into much of the same issues as the OP here but not the SPIFFS issues, thank god.

landscapejohn, 2024-06-25:
I can confirm that the fix @alba-ado added works with the LilyGO T-display-S3 using the Arduino 2 IDE. Excellent work, this should be merged to the library.

stevexyz, 2024-09-19:
Same problem here.

Inserted the line with  #include "hal/gpio_ll.h" and error disappeared.

But programs not working (thou at least display lights up).

T1j3ff, 2024-09-27:
Arduino IDE 2.3.3
Board Library: esp32 by espressif v3.0.5
Board: LiliGo T-Display 

Had this very same bug about GPIO 
Did this workaround to add  #include "hal/gpio_ll.h" in TFT_eSPI_ESP32.h

Program Compile and upload fine 
LCD back-light light up 
But nothing on the screen

IanBUK, 2024-12-28:
I'm having this same error on v 2.5.43 - and  `#include "hal/gpio_ll.h"` is already in `TFT_eSPI_ESP32.h`

This is using `#include <User_Setups/Setup50_SSD1963_Parallel.h>`. 

This helped: https://forum.arduino.cc/t/esp32-s3-sudden-compile-error/1177237/24

The solution was to step back to v2.0.11 of the esp32 board manager, from v3.0.7. I haven't stepped forwards in versions, after 2.0.11, to see at what point the function was removed, but this just worked for me.

Gyosa3, 2025-11-10:
Hi I have been using TFT_eSPI for using a TFT display with an ESP32 in the realm of ESPHOME and it compiled 2 years ago correctly. I did not bother to update a working device and now I updated all my devices to the latest now called 'ESPHome builder' firmware and I got the same errors as above because it uses expressif platform v5.x.

I tried to add the 'hal' include to my local copy of the TFT code but it does not work in the context of ESPHome as it seems, 

I see that this include was added to the later releases, but in other forums also people said it was not futureproof...

Any idea how to fix this in the longer term? I see so many people who struggle with this nowadays...