Bodmer/TFT_eSPI #2629: TFT_eSPI after the update, the library no longer works on the ESP32-2432s028

echosmart792, 2023-05-04:
This library has worked well for me so far, I used it on an ESP32-2432s028, but since yesterday, after an update, it no longer works. I thought the problem was with the microcontroller, the microcontroller has a built-in screen, I bought it here: https://www.aliexpress.com/item/1005004502250619.html
If this information helps you, to help me identify the problem why it doesn't work, the yoRadio file where the screen is configured looks like this:

`#ifndef myoptions_h
#define myoptions_h
 
#define DSP_MODEL           DSP_ILI9341
#define DSP_HSPI            true
#define LED_INVERT          true
#define TFT_CS              15
#define TFT_DC              2
#define TFT_RST             -1
#define LED_BUILTIN         17
#define BRIGHTNESS_PIN      21
#define I2S_INTERNAL            true
#define PLAYER_FORCE_MONO           true
//#define I2S_DOUT            22
//#define I2S_BCLK            16
//#define I2S_LRC             4
#define TS_MODEL            TS_MODEL_XPT2046
#define TS_SPIPINS          25, 39, 32    /* SCK, MISO, MOSI */
#define TS_CS               33
//#define IR_PIN              27
#define LIGHT_SENSOR        34
#define AUTOBACKLIGHT_MAX   1024
#endif`
I mention that if I use the LovyanGFX library, the display shows everything correctly. So the hardware is not faulty. 
However, I would prefer to use the TFT_eSPI library as I have used so far.

Can you help me solve this problem?

echosmart792, 2023-05-04:
This is my User_setup.h file in this momment.

`#define USER_SETUP_INFO "User_Setup"

#define ILI9341_DRIVER       // Generic driver for common displays
#define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only

#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 33     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
#define USE_HSPI_PORT`

Bodmer, 2023-05-04:
Which version do you have a problem with?
Which version was OK?

Run "Read_User_Setup" sketch and check the output is as you expect (output is to serial monitor only!) Post the output here.

echosmart792, 2023-05-04:
I downgraded the version, but none of them work anymore.

echosmart792, 2023-05-04:
[code]
TFT_eSPI ver = 2.5.30
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 13
SCK     = GPIO 14
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TOUCH_CS = GPIO 33

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
Touch SPI frequency   = 2.50
[/code]

echosmart792, 2023-05-04:
I found the problem. at least now I have an image on the screen as before. When updating the library, the User_Setup.h file was overwritten, and when I passed the data, I did not enter all the data.
The User_Setup.h file that works for me, I don't know if the touchscreen is ok (probably not), but the one that displays the correct image on this display is the following.

`#define USER_SETUP_INFO "User_Setup"

#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172

#define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only
#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 320 // ST7789 240 x 320

#define TFT_BL   21            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 33     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
#define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
#define USE_HSPI_PORT`

echosmart792, 2023-05-04:
Can you help me set up and check the operation of this touchscreen library?
The functional data are in the yoRadio configuration file that I published above. Maybe the settings there will help you, help me to correctly configure the tocuhscreen with this library.

closed 2023-05-04 by echosmart792