Bodmer/TFT_eSPI #2637: fillScreen() leaves a white lined block across screen

tmargot, 2023-05-06:
Hi,

First time using your library and love it. 
I am using this ILI9488 display https://www.aliexpress.com/item/1005003033841623.html which has a resolution of 320x480
My board is a TTGO-T1 with SD Card which is sharing the SPI for the LCD. 

The issue is rather easy to explain, I call fillScreen() and it does not look like it fills the entire screen leaving a lined white block across the screen which is about 1/4 (24pt Text) high. If I draw 24pt text on the display, the block remains for the remainder of the screen not covered with text. 

Here is an image of the issue: https://drive.google.com/file/d/1-fmFUsOtuxfIm1S3BhU_z9z6f7FLVYyK/view?usp=share_link
Note that the white block is relative to the rotation set, if portrait it is along the short edge of the display and the opposite for landscape which makes sense. 

I checked the code behind a fillScreen() and see that is simply uses fillRect() passing in the user configuration for the selected display driver and so thought to try just fillRect() and it worked fine. I am initialisinig TFT_eSPI via my class constructor member initializer list.

**My code pieces:**

_User_Setup.h file:_
 
`#define ILI9488_DRIVER 

.....

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

//#define TFT_MISO 19
#define TFT_MOSI 15
#define TFT_SCLK 14
#define TFT_CS   25  // Chip select control pin
#define TFT_DC    26  // Data Command control pin
#define TFT_RST   27  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

// With an ILI9163 display 27 MHz works OK.

// #define SPI_FREQUENCY   1000000
// #define SPI_FREQUENCY   5000000
// #define SPI_FREQUENCY  10000000
// #define SPI_FREQUENCY  20000000
#define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  55000000 // STM32 SPI1 only (SPI2 maximum is 27MHz)
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000
 `
_My header:_
`namespace trisM{

  class GPSProject
  {

....

 TFT_eSPI _tft{};
}
}
`

_My class (I added resolution to see if it made a difference but it didn't):_
`GPSProject::GPSProject(HardwareSerial *gpsSerial, SPIClass *sdSPI, Stream *logSerial):_tft{320,480}
{
   
    _gpsSerialIn = gpsSerial;
    _sdSPIIn = sdSPI;
    _logSerial = logSerial;
    // Initialize SPI
    _sdSPIIn->begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
    displayInit();
}

...

bool GPSProject::displayInit(){
const int16_t TFT_BG_COLOUR = TFT_BLACK; //is actually defined in settings header file but placed here for brevity 
_tft.init();
_tft.setRotation(1);
_tft.fillScreen(TFT_BG_COLOUR); // Clear screen to navy background
std::cout << "TFT Width: " << _tft.width() << std::endl;
std::cout << "TFT Height: " << _tft.height() << std::endl;
std::cout << "TFT Rotation: " << _tft.getRotation() << std::endl;
}
`

I added the logging out of the width, height and rotation to try determine the issue, I thought that the _width and _height was not being set for some reason but they checked out perfectly. 

If I call _tft.fillRect(0, 0, 480, 320, TFT_BG_COLOUR) at the end of displayInit() function, the screen successfully goes completely black.

Any ideas?

Bodmer, 2023-05-10:
It looks like poor signal integrity. Shorten connections wires, make sure there are 0V (GND) connections direct between processor and display. If colours are wrong it may be that the display is not an ILI9488 type. (Some vendors have incorrect information on the Web page.)

Bodmer, 2023-05-16:
Assumed resolved.

tmargot, 2023-05-17:
Thank you for reopening this and apologies again for missing your reply. Unfortunately not resolved. Still got the white block across the screen. I also noticed as I am displaying times that when a narrower digit is displayed, you can still see part of the previous in the remaining space. I  have shortened the wires and connected directly to Ground and V3.3. I reread your reply and see you mentioned if colours are wrong, I will check that when I get a chance and report back. 
See https://drive.google.com/file/d/11x_szM9VPEKz_oqYmMddOtWrvY4WMGpK/view?usp=share_link

Bodmer, 2023-05-17:
Run the Colour_Test sketch.

Run Read_User_Setup sketch and post the serial output here.

Post a picture showing all wiring to the board.

Try running at 10MHz SPI speed.

tmargot, 2023-05-18:
I am using Platform.io and a beginner, wanted to keep my current wiring etc. in place for testing.  Got the Colour_Test sketch working, colours look OK to me
https://drive.google.com/file/d/12QW6m59QiOas-tBUu7MmnJ4whkRcf7XP/view?usp=share_link
https://drive.google.com/file/d/12Sfxv6ayLlDS34ToYP4CET4kcXSgpqEd/view?usp=share_link

I tried running the Read_User_Setup but got an error, but ran out of time this evening and will only be able to get back to it next week as away tomorrow for the weekend:
`Building in release mode
Compiling .pio\build\ttgo-t1\libc83\TFT_eSPI\TFT_eSPI.cpp.o
In file included from .pio/libdeps/ttgo-t1/TFT_eSPI/TFT_eSPI.cpp:16:
.pio/libdeps/ttgo-t1/TFT_eSPI/TFT_eSPI.h:32:10: fatal error: SPI.h: No such file or directory`

I thought SPI was included in TTGO-T1 library, something is not hooked up correctly in this new project. 

I changed the setting to 10MHz and the large block was replaced with a single pixel line, the character remnants remain. 

https://drive.google.com/file/d/12ZRc-zYKTPcL_V7DWUmKATSyaGsXfqhd/view?usp=share_link

Here is a video at 10MHz:
https://drive.google.com/file/d/12jw-2r2sa2ut9N5L0MRlXjQOGaBGtbWb/view?usp=share_link

Thank you for your help

Bodmer, 2023-05-22:
Since the Colour_Test is working fine then this is likely be a problem with the sketch. For example: a hardware resource conflict, badfont files or memory corruption/resource limits.

Provide a complete sketch that demonstrates the problem.

tmargot, 2023-05-25:
Thank you for the suggestion, I still have not had a chance to get back to
this, life just gets in the way :)
On the weekend I hope to create a new sketch with code extracted from my
project to do some further testing. I will report back on my findings,
thank you for your help and patience.

closed 2023-06-05 by Bodmer

Bodmer, 2023-06-05:
Assumed resolved.