Bodmer/TFT_eSPI #2646: Problema ao compilar.  Versão TFT_eSPI  2.5.30  e ESP32 arduino 2.0.9

cdm621, 2023-05-10:
In file included from c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:0:
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA(bool)':
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: sorry, unimplemented: non-trivial designated initializers not supported
   };
   ^
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: sorry, unimplemented: non-trivial designated initializers not supported
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: sorry, unimplemented: non-trivial designated initializers not supported
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: sorry, unimplemented: non-trivial designated initializers not supported
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: sorry, unimplemented: non-trivial designated initializers not supported
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: sorry, unimplemented: non-trivial designated initializers not supported
c:\Users\damia.DAMIAN-LENOVO\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: sorry, unimplemented: non-trivial designated initializers not supported

exit status 1

Compilation error: exit status 1

cdm621, 2023-05-10:
Depois de atualizar a versão do IDE para 2.0.9 começou esse problema, se alguém consegue me ajudar agradeço.

Bodmer, 2023-05-10:
Which version of the ESP32 board package are you using?

cdm621, 2023-05-10:
estava com o último pacote 2.0.9.
hoje consegui solucionar o problema instalando versão anterior 2.0.7
Obrigado pela atenção!

Bodmer, 2023-05-21:
I do not get the error with 2.0.9 and an ESP32 so I think the library has probebly been updated. Try the latest master copy.

closed 2023-05-21 by Bodmer